mavlink/qgroundcontrol #9419: invalid value prevents mission with landing pattern created in QGC 4.1.1 on macOS from uploading to AP 4.0.8

rtrevellyan, 2021-02-07:
When I create a mission in QGC 4.1.1 on macOS it won't upload to ArduPlane 4.0.8. The response is that there is an invalid value. This seems to be related to landing patterns.
----

## Expected Behavior
I expect to be able to upload a mission with a landing pattern that I've created in QGC to the current stable release of ArduPlane.

## Current Behavior
Mission upload fails with a message about Param 5 invalid value.

## Steps to Reproduce:
1. Create a mission that includes a landing pattern.
2. Connect to vehicle running ArduPlane 4.0.8 and attempt to upload the mission.
3. Notice the error message about a missing required parameter. "Mission transfer failed. Error: Param 5 invalid value. Item #xx Command: Stop image capture Value: nan"
4. The upload fails.
5. Delete the landing pattern and replace it with an RTL.
6. Upload succeeds.

## System Information
- Operating System: macOS Catalina 10.15.7
- QGC Version: 4.1.1
- QGC build: stable
- Flight Controller: Matek H743-Wing
- Autopilot (with version): ArduPlane 4.0.8

## Detailed Description
The mission also won't load in QGC 4.0.11. In that case the error message is "Error loading Plan file (<path>.plan). Mission: The following required keys are missing: loiterCoordinate"

DonLakeFlyer, 2021-02-22:
This doesn't reproduce for me. My guess is that you created this Plan while disconnected from the vehicle. And you didn't change the Vehicle Info in Mission Start to ArduPilot firmware. If you are connected to an ArduPilot vehicle this will happen automatically. ArduPilot does not support the Stop Image Capture command which is part of Landing Pattern. But that would only be part of the pattern it the Plan was created for PX4 Firmware.

If you can reproduce with the Vehicle Info set to ArduPilot firmware I will re-open and creating a new plan with those settings I will re-open.

rtrevellyan, 2021-02-23:
I can reproduce with the vehicle info set to ArduPilot Fixed Wing. The problem is a little more subtle than I originally thought. A newly created plan can be uploaded to the vehicle without error. However, after saving the plan and then reloading it, it will no longer upload. There also seems to be a related problem when loading a plan saved from a previous version of QGC.

DonLakeFlyer, 2021-02-23:
Can you attach the plan here that gives you the error?

rtrevellyan, 2021-02-23:
Here it is.
[trivial.plan.zip](https://github.com/mavlink/qgroundcontrol/files/6026823/trivial.plan.zip)

cjzilla, 2021-06-22:
Bump - above issue still exists

DonLakeFlyer, 2021-06-24:
Fixed. 

Landing Patterns saved from daily builds without this fix will be wrong. They look like this:
```
            {
                "altitudesAreRelative": true,
                "complexItemType": "fwLandingPattern",
                "landCoordinate": [
                    47.72760037,
                    -122.20156905,
                    0
                ],
                "landingApproachCoordinate": [
                    47.72827474798988,
                    -122.19648088485897,
                    40
                ],
                "loiterClockwise": true,
                "loiterRadius": 75,
                "stopTakingPhotos": true,
                "stopVideoPhotos": false,
                "type": "ComplexItem",
                "valueSetIsDistance": false,
                "version": 2
            }
```

You can fix them by hand by changing stopTakingPhotos to false. Or you can use a new build to save a new plan.

PR #9724 (2021-06-25): FW Landing Pattern: Fix json key name
Fix #9419
closed 2021-06-24 by DonLakeFlyer