mavlink/qgroundcontrol #9423: Stable_V4.1: "Apply Preset" fails with error, missing keys

dakejahl, 2021-02-08:
When trying to "Apply Preset", I get an error about missing keys in the JSON object. I looked at the .json file for the preset, and it indeed has these keys it says are missing. This happens even after creating a fresh new preset from a Basic Survey as seen in the image below. I just created a preset called "test", saved it as a new preset, and then tried to load it. 

[image]

dakejahl, 2021-02-17:
@patrickelectric 

Looks like this change broke it
https://github.com/mavlink/qgroundcontrol/commit/d111e78497c1f65d3f0f18c6250d14b852c83651

patrickelectric, 2021-02-17:
Fix for reporting this, should be fixed on #9447

PR #9447 (2021-02-18): MissiongManager: ComplexMissionItem: Save preset as CborValue variant
Fix #9423
commit 6bdb7c6b (2021-02-18): MissiongManager: ComplexMissionItem: Save preset as CborValue variant

Fix #9423
PR #9455 (2021-02-19): MissiongManager: ComplexMissionItem: Save preset as CborValue variant
Fix #9423

Requested cherry-pick
commit 85d71ca5 (2021-02-19): MissiongManager: ComplexMissionItem: Save preset as CborValue variant

Fix #9423
closed 2021-02-18 by DonLakeFlyer