mavlink/qgroundcontrol #9425: Landing approach shows red in Fly view, but orange in Plan view

Antiheavy, 2021-02-09:
Is the red intentional to indicate caution?  If so, it seem like we should use something different that the same red color used to indicate terrain intersection in the Plan screen.  Maybe orange triangle with red boarder or something?   I also suspect this might just be a bug...

Fly view:
[image]

Plan view:
[image]

Mission file:
[red-fly-view-landing.zip](https://github.com/mavlink/qgroundcontrol/files/5952642/red-fly-view-landing.zip)

DonLakeFlyer, 2021-02-16:
Most likely just a bug

Antiheavy, 2021-02-27:
This bug is still there in the latest stable.

DonLakeFlyer, 2021-03-02:
I'll associate that with #9495. Once that gets taken care of we'll see if it still happens.

Antiheavy, 2021-03-20:
this is still happening with the latest test build on stable with attempted fixes for #9495 
[image]

GIF:
[image]

Mission file:
[landing-approach-red.zip](https://github.com/mavlink/qgroundcontrol/files/6175773/landing-approach-red.zip)

DonLakeFlyer, 2021-03-25:
Should be fixed now

Antiheavy, 2021-03-26:
seems fixed based on testing latest daily!  thanks.

DonLakeFlyer, 2021-03-26:
@Antiheavy Thanks so much for helping with testing as always.

blackbareng, 2022-03-01:
We are having similar issues as mentioned above when planning a mission for a small fixed wing. 

We’ve planned this exact mission several times before, but in the last few days, the landing pattern is yellow when planning, but turns red when uploaded. The glide slope is 6 degrees. No different than it’s ever been. 

Is this the same bug discussed above?

DonLakeFlyer, 2022-03-02:
Save the plan that isn't working and attach it here

blackbareng, 2022-03-02:
Hello Don,

The plan is attached below, as well as two images from our controller.

The plan has been flown several times before, but now, is currently going
red upon mission upload. During the planning phase, the landing pattern is
yellow as normal, but once uploaded it becomes red.

If we change the location of the launch point on the screen, the pattern
immediately returns to normal - but it will not accept the launch location
as it normally has.

Additionally, the launch point on the controller screen keeps jumping to
the exact location of the craft, rather than staying where we will be
launching the craft.

*CALLIE GROTH*

*BLACKBAR ENGINEERING*
c 520.226.6302
www.blackbarengineering.com