mavlink/qgroundcontrol #9428: Sensor Status not scrollable therefore force-arm button not reachable

0815Creeper, 2021-02-10:
Android:
tested with regular playstore version and daily test build playstore version

## Expected Behavior
The sensor status list should be scrollable, so that the Force-Arm button at the bottem (I think there should be one according to https://github.com/mavlink/qgroundcontrol/pull/8993) becomes visible

## Current Behavior
Scrolling not possible: some sensor states not visible and force-arm button not visible.

## Steps to Reproduce:
1. open App on Android-Phone with regular screen size (1080x1920-423ppi in my case), connect to any drone with a few sensors...

## System Information
- Operating System: Android 8.1.0
- QGC Version: v4.1.1 64bit
- QGC build: playstore version
- Flight Controller: Matek H743wing
- Autopilot (with version): Ardupilot v4.0.8
 
## Detailed Description
I do not know what would remain to be clarifyed.

## Log Files and Screenshots
[image]

DonLakeFlyer, 2021-02-25:
Yup

closed 2021-03-02 by DonLakeFlyer

0815Creeper, 2021-07-04:
didnt make it to production, right?

DonLakeFlyer, 2021-07-07:
It is bu Android store is not updating. APKs are available from the web site

PR #9489 (2021-02-26): Sensor Status: Correct sizing, flickable
* Fix max sizing
* Make is flickable
* Move arm/disarm button to top for usability
* Fix #9428

[image]