mavlink/qgroundcontrol #9433: UI Issues in QGC v4.1 and recent.

sunny52266, 2021-02-12:
1. Certain UI features not visible until the application is scaled to around 70 percent on android phone. My current phone is oneplus 6. 

2. The translucent nature of the main dock causes readability issues in sunlight specially when the satellite imagery is turned on and bare ground is in the background. For example: [image]

Davidsastresas, 2021-02-13:
fixed in  #9436

DonLakeFlyer, 2021-02-16:
I think this screenshot is showing more bugs than just the visibility problem. You can see how the coloring of the text versus the labels differs. I need to take a look at that to see if there is an additional problem.

DonLakeFlyer, 2021-02-17:
Indoor theme looks like crap as well over certain maps. I think in both cases the values panel needs to be opaque.

Davidsastresas, 2021-02-17:
That makes sense. What do you think is the cause of the other bug? it seems like the bold feature of the digits is not what it should right?

DonLakeFlyer, 2021-03-02:
> 1. Certain UI features not visible until the application is scaled to around 70 percent on android phone. My current phone is oneplus 6.

@sunny52266 What else is missing?

sunny52266, 2021-03-02:
1. Some times the back button does not exit the application. 

2. When use vertical instrument panel is checked, the instrument panel disappears. 

3. There is no option for trigger camera now as there was in previous versions. 

4. Second GPS info does not appear in the gps parameters selection in the parameters dock type thingy on the main display that needed to be opaque. 

These are the ones from the top of my head.

DonLakeFlyer, 2021-03-02:
> 2\. When use vertical instrument panel is checked, the instrument panel disappears.

Fixed. Will show up when a new stable is marked.

> 3\. There is no option for trigger camera now as there was in previous versions.

This should be fixed as well. Will show up when new stable is marked.

> 4\. Second GPS info does not appear in the gps parameters selection in the parameters dock type thingy on the main display that needed to be opaque.

This was not supported in previous versions. Support was just added and is supported in Daily builds.

sunny52266, 2021-03-03:
@DonLakeFlyer Must be good then. Shall I close this issue now?

closed 2021-03-03 by sunny52266