mavlink/qgroundcontrol #9435: Option to display values for GPS2 instrument panel (the selectable one)

pompecukor, 2021-02-13:
On the user configurable display values, would be nice to be able to also show GPS2 (count and HDOP).
Right now only possible to show for GPS1. one, but that is already show on the horizontal bar at the top,

For us this is mainly useful for when we are testing the aircraft. On QGC there is just no was to make GPS2 information to show.
BTW same goes for secondary Airspeed sensor. 

Thank you.

Davidsastresas, 2021-02-13:
Partially fixed in #9437. For the airspeed matter I think we need development from mavlink and autopilot firmware as well, as I couldn't see where is currently sent the second airspeed sensor info.

Davidsastresas, 2021-02-14:
Okey, apparently Airspeed 2 is sent right now from Ardupilot by a "named float" mavlink message. This is a workaround until we have support for it from mavlink, so we should really open an issue in mavlink, and when we have proper support in mavlink for this add it to QGC.

eduardkieser, 2023-02-07:
Hi @Davidsastresas, I have a related question. I have implemented a sensor driver in lua for an unsupported temperature sensor. I would like to view the values of that sensor in the "Instrument Panel". The message is being sent as a `named_float`, and on mission planner I can configure the quick view to display the value. Is there any way that I can view named floats in the QGC gui?

Davidsastresas, 2023-02-07:
I don't think we have support for that right now, but it can be added. Can you open a separate issue for that matter? thanks!

eduardkieser, 2023-02-08:
sure thing. [10564](https://github.com/mavlink/qgroundcontrol/issues/10564).

closed 2026-10-04 by github-actions[bot]