mavlink/qgroundcontrol #9440: RTSP video from specific video cam not displayed since 4.1.0

tzuriel-spearuav, 2021-02-15:
I'm working with Nextvision video camera, and since v4.1.0 video is not displayed.
The log has the following error :
```
[D] at C:\projects\qgroundcontrol\src\QGCApplication.cc:287 - "Settings location "C:/Users/TzurielLampner/AppData/Roaming/QGroundControl.org/QGroundControl.ini" Is writable?: true"
[D] at C:\projects\qgroundcontrol\src\QGCLoggingCategory.cc:118 - "Filter rules "*Log.debug=false\nGStreamerAPILog.debug=true\n""
[E] at ../gst/gstelementfactory.c:467 - "(NULL) no such element factory "qmlglsink"!"
[D] at C:\projects\qgroundcontrol\src\QGCApplication.cc:390 - "System reported locale: QLocale(English, Default, Israel) ; Name "en_IL" ; Preffered (used in maps):  "en-US""
[E] at C:\projects\qgroundcontrol\src\QGCApplication.cc:482 - "Error loading source localization for "en_IL""
[E] at C:\projects\qgroundcontrol\src\QGCApplication.cc:487 - "Error loading json localization for "en_IL""
[E] at :0 - "serialnmea: No known GPS device found. Specify the COM port via QT_NMEA_SERIAL_PORT."
[D] at C:\projects\qgroundcontrol\src\Vehicle\MAVLinkLogManager.cc:360 - "MAVLink logs directory: "C:/Users/TzurielLampner/OneDrive - SpearUAV/Documents/QGroundControl/Logs""
[D] at C:\projects\qgroundcontrol\src\QtLocationPlugin\QGCMapEngine.cpp:164 - "Map Cache in: "C:/Users/TzurielLampner/AppData/Local/cache/QGCMapCache300" / "qgcMapCache.db""
[D] at qrc:/qml/MainRootWindow.qml:51 - "QGCCorePlugin(0x238f83d1180) []"
[E] at qrc:/qml/MainRootWindow.qml:412 - "qrc:/qml/MainRootWindow.qml:412:13: QML ColumnLayout: Possible anchor loop detected on vertical anchor."
[E] at ../gst/videoparsers/gsth264parse.c:1492 - "<h264parse0> broken/invalid nal Type: 1 Slice, Size: 8820 will be dropped"
[E] at ../gst/videoparsers/gsth264parse.c:1492 - "<h264parse0> broken/invalid nal Type: 1 Slice, Size: 8559 will be dropped"
[E] at ../gst/videoparsers/gsth264parse.c:3450 - "<h264parse1> H.264 AVC caps, but no codec_data"
[E] at ../gst/videoparsers/gsth264parse.c:3476 - "<h264parse1> refused caps video/x-h264, stream-format=(string)avc, alignment=(string)au, width=(int)1280, height=(int)720, framerate=(fraction)0/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, profile=(string)constrained-baseline, level=(string)3.1"
[E] at ../gst/videoparsers/gsth264parse.c:3450 - "<h264parse1> H.264 AVC caps, but no codec_data"
[E] at ../gst/videoparsers/gsth264parse.c:3476 - "<h264parse1> refused caps video/x-h264, stream-format=(string)avc, alignment=(string)au, width=(int)1280, height=(int)720, framerate=(fraction)0/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, profile=(string)constrained-baseline, level=(string)3.1"
[E] at ../gst/videoparsers/gsth264parse.c:3450 - "<h264parse1> H.264 AVC caps, but no codec_data"
[E] at ../gst/videoparsers/gsth264parse.c:3476 - "<h264parse1> refused caps video/x-h264, stream-format=(string)avc, alignment=(string)au, width=(int)1280, height=(int)720, framerate=(fraction)0/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, profile=(string)constrained-baseline, level=(string)3.1"
[E] at ../gst/videoparsers/gsth264parse.c:3450 - "<h264parse1> H.264 AVC caps, but no codec_data"
[E] at ../gst/videoparsers/gsth264parse.c:3476 - "<h264parse1> refused caps video/x-h264, stream-format=(string)avc, alignment=(string)au, width=(int)1280, height=(int)720, framerate=(fraction)0/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, profile=(string)constrained-baseline, level=(string)3.1"
[E] at ../gst/videoparsers/gsth264parse.c:3450 - "<h264parse1> H.264 AVC caps, but no codec_data"
[E] at ../gst/videoparsers/gsth264parse.c:3476 - "<h264parse1> refused caps video/x-h264, stream-format=(string)avc, alignment=(string)au, width=(int)1280, height=(int)720, framerate=(fraction)0/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, profile=(string)constrained-baseline, level=(string)3.1"
[E] at ../gst/videoparsers/gsth264parse.c:3450 - "<h264parse1> H.264 AVC caps, but no codec_data"
[E] at ../gst/videoparsers/gsth264parse.c:3476 - "<h264parse1> refused caps video/x-h264, stream-format=(string)avc, alignment=(string)au, width=(int)1280, height=(int)720, framerate=(fraction)0/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, profile=(string)constrained-baseline, level=(string)3.1"
[E] at ../libs/gst/base/gstbasesrc.c:3127 - "<udpsrc1> error: Internal data stream error."
[E] at ../libs/gst/base/gstbasesrc.c:3127 - "<udpsrc1> error: streaming stopped, reason not-negotiated (-4)"
[!] at C:\projects\qgroundcontrol\src\VideoReceiver\GstVideoReceiver.cc:1305 - "GStreamer error: Internal data stream error."
[E] at ../gst/rtsp/gstrtspsrc.c:6469 - "<source> server closed connection"
[E] at ../gst/rtsp/gstrtspsrc.c:6469 - "<source> server closed connection"
[E] at ../gst/rtsp/gstrtspsrc.c:6506 - "<source> error: Could not receive message. (Received end-of-file)"
[E] at ../gst/rtsp/gstrtspsrc.c:8215 - "<source> error: Could not send message. (Received end-of-file)"
```

Running v4.0.11 works perfectly.
Trying to revet changes in GstVideoReceiver.cc and .h (from mavlink/qgroundcontrol/pull/9213, mainly going back to 'decodebin' instead of 'decodebin3', but also restored back _autoplugQueryXX functions), seems to solve the problem - video is now shown.
@patrickelectric can you help ?

LorenzMeier, 2021-02-16:
@DanielePettenuzzo Could you please look into this with @andrewvoznytsa to at least provide some basic indications?

patrickelectric, 2021-02-17:
I just tested in master branch and it appears to be working:
[image]
You can always test with rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov

I'm not familiar with this camera, but how the stream is created ? Is provided directly by the camera or is there a software in the middle ?
Can you test with another software such as VLC ?

tzuriel-spearuav, 2021-02-17:
I cannot see even the bunny video in clean v4.1.0 or v4.1.1 release installation (didnt try master latest).
I am able to see this video in both VLC and gstreamer command line, so dont think this is related to firewall (btw, gstreamer command line with default decodebin: ```gst-launch-1.0.exe -v rtspsrc location=rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov latency=10 ! decodebin ! autovideosink```)

Tried all the video decoder priority with no success. Though for both DirectX3D11 and Software decoders I get some console error:
```
[D] at C:\projects\qgroundcontrol\src\QGCApplication.cc:287 - "Settings location "C:/Users/Tzuriel/AppData/Roaming/QGroundControl.org/QGroundControl.ini" Is writable?: true"
[D] at C:\projects\qgroundcontrol\src\QGCLoggingCategory.cc:118 - "Filter rules "*Log.debug=false\nGStreamerAPILog.debug=true\n""
[E] at ../gst/gstelementfactory.c:467 - "(NULL) no such element factory "qmlglsink"!"
[D] at C:\projects\qgroundcontrol\src\QGCApplication.cc:390 - "System reported locale: QLocale(English, Default, Israel) ; Name "en_IL" ; Preffered (used in maps):  "en-US""
[E] at C:\projects\qgroundcontrol\src\QGCApplication.cc:482 - "Error loading source localization for "en_IL""
[E] at C:\projects\qgroundcontrol\src\QGCApplication.cc:487 - "Error loading json localization for "en_IL""
[D] at C:\projects\qgroundcontrol\src\Vehicle\MAVLinkLogManager.cc:360 - "MAVLink logs directory: "C:/Users/Tzuriel/OneDrive - SpearUAV/Documents/QGroundControl/Logs""
[D] at C:\projects\qgroundcontrol\src\QtLocationPlugin\QGCMapEngine.cpp:164 - "Map Cache in: "C:/Users/Tzuriel/AppData/Local/cache/QGCMapCache300" / "qgcMapCache.db""
[D] at qrc:/qml/MainRootWindow.qml:51 - "QGCCorePlugin(0x17f196a8920) []"
[E] at qrc:/qml/MainRootWindow.qml:412 - "qrc:/qml/MainRootWindow.qml:412:13: QML ColumnLayout: Possible anchor loop detected on vertical anchor."
[D] at C:\projects\qgroundcontrol\src\comm\UDPLink.cc:205 - "Adding target QHostAddress("127.0.0.1") 53199"
[D] at C:\projects\qgroundcontrol\src\comm\MAVLinkProtocol.cc:214 - "Switching outbound to mavlink 2.0 due to incoming mavlink 2.0 packet: 0x7ff72d1f8b68 1 2"
[D] at C:\projects\qgroundcontrol\src\comm\MAVLinkProtocol.cc:463 - "Temp log "C:/Users/Tzuriel/AppData/Local/Temp/FlightData766340.mavlink""
[E] at :0 - "libpng warning: iCCP: known incorrect sRGB profile"
[D] at C:\projects\qgroundcontrol\src\QGCApplication.cc:899 - ""v4.1.1""
[D] at C:\projects\qgroundcontrol\src\Vehicle\Vehicle.cc:2068 - "_parametersReady true"
[E] at ../gst/rtsp/gstrtspsrc.c:5856 - "<source> warning: Could not receive any UDP packets for 5.0000 seconds, maybe your firewall is blocking it. Retrying using a tcp connection."
[E] at ../gst/rtsp/gstrtspsrc.c:6410 - "<source> receive interrupted"
[E] at ../gst/rtsp/gstrtspsrc.c:6508 - "<source> receive interrupted"
[E] at ../gst/rtsp/gstrtspsrc.c:8776 - "<source> PLAY interrupted"
[E] at ../gst/rtsp/gstrtspsrc.c:6410 - "<source> receive interrupted"
[E] at ../gst/rtsp/gstrtspsrc.c:6508 - "<source> receive interrupted"
[E] at ../gst/rtsp/gstrtspsrc.c:8905 - "<source> PAUSE interrupted"
[E] at ../gst/rtsp/gstrtspsrc.c:6086 - "<source> Timed out waiting for TEARDOWN to be processed."
[E] at ../gst/rtsp/gstrtspsrc.c:6410 - "<source> receive interrupted"
[E] at ../gst/rtsp/gstrtspsrc.c:6508 - "<source> receive interrupted"
[E] at ../gst/rtsp/gstrtspsrc.c:8217 - "<source> TEARDOWN interrupted"
[E] at ../gst/rtsp/gstrtspsrc.c:6410 - "<source> receive interrupted"
[E] at ../gst/rtsp/gstrtspsrc.c:6508 - "<source> receive interrupted"
[E] at ../gst/rtsp/gstrtspsrc.c:8905 - "<source> PAUSE interrupted"
[E] at ../gst/rtsp/gstrtspsrc.c:6086 - "<source> Timed out waiting for TEARDOWN to be processed."
[E] at ../gst/rtsp/gstrtspsrc.c:6410 - "<source> receive interrupted"
[E] at ../gst/rtsp/gstrtspsrc.c:6508 - "<source> receive interrupted"
[E] at ../gst/rtsp/gstrtspsrc.c:8217 - "<source> TEARDOWN interrupted"
[E] at ../gst/rtsp/gstrtspsrc.c:6410 - "<source> receive interrupted"
[E] at ../gst/rtsp/gstrtspsrc.c:6508 - "<source> receive interrupted"
[E] at ../gst/rtsp/gstrtspsrc.c:8905 - "<source> PAUSE interrupted"
[E] at ../gst/rtsp/gstrtspsrc.c:6086 - "<source> Timed out waiting for TEARDOWN to be processed."
```

Have a hunch this may be a windows specific issue, or related to video card properties (I have an Intel UHD Graphics adapter in my Dell laptop).

DanielePettenuzzo, 2021-02-20:
I also tested v4.1.0 and master on linux and it works fine for me. I posted some tests here: https://github.com/mavlink/qgroundcontrol/pull/9449#issuecomment-782733076

I haven't tested on Windows yet. @andrewvoznytsa and I will spend some time on Monday to test on windows.

andrewvoznytsa, 2021-02-22:
@tzuriel-spearuav I've tested https://s3-us-west-2.amazonaws.com/qgroundcontrol/latest/QGroundControl-installer.exe (v.4.1.1) on Microsoft Windows [Version 10.0.17134.1246] running under VMware on Linux host with enabled 3D acceleration (recent AMD CPU and nVidia VGA) - RTSP and UDP (H.264) streaming work for me.

Could you please capture your RTSP network traffic? You could use wireshark (see https://www.wireshark.org/) to do that.

andrewvoznytsa, 2021-02-22:
@tzuriel-spearuav tried with your public RTSP bunny video - reproduced on Windows. Will check what's wrong. But if my stream works (H.264 1080p, no audio) then I think it can also work on your side. Are you sure that you need such small resolution video and audio? Perhaps it would be good idea to increase resolution and drop audio.

tzuriel-spearuav, 2021-02-22:
@andrewvoznytsa attached is the rtsp capture.
[rtsp_capture.zip](https://github.com/mavlink/qgroundcontrol/files/6021343/rtsp_capture.zip)

The bunny video is just a public sample, I'm getting same results with my video-camera feed which doesn't have audio in it, and its resolution is 1280x720

andrewvoznytsa, 2021-02-24:
Have compiled v4.1.1 on my side - works perfectly. I think it is kind of misconfiguration and not the video bug.

[image]

tzuriel-spearuav, 2021-02-28:
Indeed after testing multiple times I am able to view the bunny video occasionally.
From what I gather the problem only happens if you change the RTSP address **after** QGC has started. If the RTSP address is configured pre-running, the video seems to show normally

andrewvoznytsa, 2021-02-28:
Just do double check - you start QGC with 'Video disabled', then switch it to 'RTSP' and enter URL - in that case you have no video. If you restart QGC after that then you see that video without any issues? Sounds strange to me - from video receiver perspective these use cases are exactly the same. But I'll check it.

tzuriel-spearuav, 2021-03-01:
Peculiar as it sounds, that's exactly what I see.
With a new error -
`VideoReceiverLog: gst_element_link_pads() failed`

andrewvoznytsa, 2021-03-04:
What can I say - works for me in both cases. Perhaps it is hardware related - some element specific to your environment does not always work correctly. Hard to guess. I suggest to try at different systems (or conditions) and see what changes - I need to reproduce it here to fix.

xdwgood, 2021-03-09:
@andrewvoznytsa  I try use rtsp video stream on qgc.

v4.0.11 work normally(have video display)

but when I try v4.1.1,it not work(no video display)

I use ubtun18.04 and gstreamer version is 1.14.5.

andrewvoznytsa, 2021-03-10:
@tzuriel-spearuav I re-tested this issue today on Linux (Ubuntu 20.10 + gstreamer 1.18.0) and to my surprise I was able to reproduce it exactly like you described (but it always worked for me on Linux). So I'll try to fix. @xdwgood FYI

patrickelectric, 2021-03-18:
Can someone confirm that #9520 fix the problem ? Ping @tzuriel-spearuav

tzuriel-spearuav, 2021-03-21:
Sorry @andrewvoznytsa @patrickelectric, tested the commit and it doesn't seem to solve my problem

xdwgood, 2021-03-23:
on Linux (Ubuntu 20.04 + gstreamer 1.16.2),rtsp video display normally

leepadc, 2021-08-09:
@LorenzMeier Hello,  Back in February you closed 9240 and moved to this thread.  However, the original problem from 9240 has not been solved.  We continue to not be able to get video from the Airborne Innovations Picoraptor.  Jon0x0 posted a working pipeline in thread 9240 after it was closed.  Is there a way to get that pipeline tested for inclusion in future builds, or is there an alternate pipeline against which we should test the Picoraptor?  I do not have the resources to modify or build a solution, but I am more than happy to test.  Thanks. -Doug

geomav88, 2021-12-24:
Same issue here with my SkyViper Journey. RTSP streaming is working with 4.0.6 on either Android or Ubuntu. No streaming after 4.1.0 onward.

fredowski, 2022-09-08:
@leepadc, @gemomav88: I had very similar problems with the RTSP bunny example in PR #10396 and I figured out that this is maybe due to a timeout. Could you try if you can connect with one of the build artefacts from #10396? The linux AppImage is here: https://github.com/mavlink/qgroundcontrol/actions/runs/3017309574

fredowski, 2022-09-08:
@tzuriel-spearuav, @geomav88 : Could you maybe try if this is fixed in PR #10396 ?

tzuriel-spearuav, 2022-09-11:
@fredowski looks pretty good :ok_hand: on same laptop, both when RTSP address is pre-configured and re-configured after app starts. Although I could not currently get a Nextvision type camera to test

PR #10409 (2022-10-04): Video: increase timeout for RTSP video streaming (#9440)
In #9440 the video streaming failed for some cameras and the bunny example RTSP streaming movie. This worked in 4.0.6 and it did not work anymore in 4.1.0. The same unstable behaviour is observed in #10396.

The gstreamer rtsp source will switch to tcp when no data is received via udp after 5 seconds. The default timeout for the gstreamer video receiver is 2 seconds resulting in a permanent teardown and restart until eventually some data is received. As far as I understood the code 4.0.6 did not have a timeout which is in 4.1.0. I change [... cut]
commit 7c3f4581 (2022-10-04): Video: increase timeout for RTSP video streaming (#9440)

In #9440 the video streaming failed for some cameras and the
bunny example RTSP streaming movie. This worked in 4.0.6 and it
did not work anymore in 4.1.0. The same unstable behaviour is
observed in #10396.

The gstreamer rtsp source will
switch to tcp when no data is received via udp after 5 seconds.
The default timeout for the gstreamer video receiver is 2 seconds
resulting in a permanent teardown and restart until
eventually some data is received. As far as I understood the code
4.0.6 did not have a timeout which is in 4.1.0. I chang [... cut]
closed 2022-10-04 by mrpollo