mavlink/qgroundcontrol #9441: Pre-Flight Checklist regression

dakejahl, 2021-02-16:
The Preflight Checklist has regressed in a few ways since Stable 4.0

**1**

The logic for popping up the checklist when "use checklist" and "enforce checklist" are selected, has been killed without being removed.
Half of the code in this file now does nothing
https://github.com/mavlink/qgroundcontrol/blob/master/src/FlightDisplay/FlyViewPreFlightChecklistPopup.qml#L24-L55
https://github.com/mavlink/qgroundcontrol/blob/master/src/FlightDisplay/FlyViewPreFlightChecklistPopup.qml#L62-L71

It could be refactored down to just this -- the code for the timer actually spits out errors on the console since it is incorrect with regard to this new implementation for the checklist.
```
QGCPopupDialog {
        id:         _root
        title:      qsTr("Pre-Flight Checklist")
        buttons:    StandardButton.Close

        Loader {
            id:     checkList
            source: QGroundControl.corePlugin.options.preFlightChecklistUrl
        }
    }
```

**2**
The checklist now resets itself when it is closed. I think this is because the item is now reloading each time the checklist is opened via the new base classed QGCPopupDialog.
https://github.com/mavlink/qgroundcontrol/blob/master/src/ui/MainRootWindow.qml#L220-L224

**3**
Checklist cannot be easily re-anchored in the new base-class architecture, which was an easy thing to do with the previous implementation.

dakejahl, 2021-02-16:
I would love to fix all of these issues -- I just need a little bit of guidance on how to do it within the new architecture.

Perhaps the checklist can be reinstated as a Popup directly within the FlyViewWidgetLayer rather than being created via 
```
onDisplayPreFlightChecklist: mainWindow.showPopupDialogFromComponent(preFlightChecklistPopup)
```
To me that seems like the simplest solution, and then I can add back the logic for displaying it via the timer.

DonLakeFlyer, 2021-02-16:
Wow, I managed to royally screw that up. I changed it to use the standard dialog mechanism so that it matches all the other ui. Not sure if that was when the auto-popup stuff broke or not. But that broke the remembering of status. I'd still like to keep it using the standard dialog mechanism for ui consistency. The fix for remembering is to keep the preflight state around as a global object like it use to be. It should be easy enough to fix the auto-popup stuff using the new dialog mechanism. Is there some complicated problem with that? I can fix that state problem if you fix the popup stuff.

DonLakeFlyer, 2021-02-16:
Also would like to wait to hear back from @dogmaphobic on the move that was made to make this a Popup instead of a DropPanel. I think it's worthwhile considering moving it back to DropPanel due to user model problems.

dakejahl, 2021-02-17:
I agree with you, let's see what Gus thinks

dogmaphobic, 2021-02-18:
I haven't touched any of this in well over a year. The file referenced above was created by @DonLakeFlyer.

dakejahl, 2021-02-18:
> I changed it to use the standard dialog mechanism so that it matches all the other ui

I wonder if we should revert this then? Since this mechanism is stateless (it creates the component every time) and it also is hard coded to anchor things to the center of the screen in the popup container. Then there's the control over when it pops up or closes... since all of that extra logic is unnecessary for other popup UI items, it seems like the checklist is better off being its own thing.

DonLakeFlyer, 2021-02-18:
@dogmaphobic The question for you relates to an Issue in the Sentera issue list not really the problem show here. Those are mainly bugs I introduced. I pinged you on Slack about it. It's about the reasoning as to why the checklist needed to be moved away from using a DropPanel and changed to a Popup. I believe that was something you changed ages ago.

DonLakeFlyer, 2021-02-18:
The issue with a Popup versus a DropPanel is that it covers too much of the center of the mission area of the Fly View. It's common for a custom pre-flight checklist to request the user to review the mission being flown. If the checklist itself covers most of the mission it makes that step a PITA.

DonLakeFlyer, 2021-03-02:
> I wonder if we should revert this then?

Yes, I think that's a good first step. I would also eventually like to consider moving back to a DropPanel but that can wait.

closed 2024-09-22 by dakejahl