mavlink/qgroundcontrol #9443: Mission transfer failed. Error: Param 1 invalid value. Item #0 Command: Takeoff Value: 15

96moustafa, 2021-02-16:
## Expected Behavior
After creating a new mission in QGC, it shall be uploaded to px4.

## Current Behavior
I get the following error:
**Mission transfer failed. Error: Param 1 invalid value. Item #0 Command: Takeoff Value: 15**

I also get the following warning;
**Takeoff rejected, remove deprecated minimum pitch after pressing the upload button.**

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. build px4_sitl: HEADLESS=1 make px4_sitl_rtps gazebo
2. create a mission in QGC UI.
3. Press the upload button.

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 20.04
- QGC Version: 4.1.1
- QGC build: stable app image
- Autopilot (with version): PX4 master

DonLakeFlyer, 2021-02-16:
Please attach the plan file which is failing

96moustafa, 2021-02-16:
[mission_1.txt](https://github.com/mavlink/qgroundcontrol/files/5991460/mission_1.txt)
This is an example of the file that failed to be uploaded. It fails with anything created from QGC mission UI.

gendebeat, 2021-02-19:
i have the same issue.

    build px4_sitl: make px4_sitl jmavsim
    create a mission in QGC UI DAILY

DonLakeFlyer, 2021-02-19:
Two things going on here:
* QGC sets NAV_TAKEOFF.Pitch=15 even for vehicles other than fixed wings
* PX4 Firmware seems to have deprecated NAV_TAKEOFF.Pitch usage entirely. It now fails uploads for values other than 0. This in turn creates a situation making it next to impossible to handle back/forward compat of QGC running against different versions of firmware.

DonLakeFlyer, 2021-02-19:
@LorenzMeier This may need discussion in a call to work out how to handle this in a back compat way.

DonLakeFlyer, 2021-02-19:
> * QGC sets NAV_TAKEOFF.Pitch=15 even for vehicles other than fixed wings

For now, I'm going to fix this but leave setting Pitch=15 for fixed wings. This leaves QGC broken for fixed wings against PX4 master but it seems like the only safe thing to do in order to preserver back compat of QGC with PX4 Stable.

DonLakeFlyer, 2021-02-19:
The only thing I can think of for now is to change firmware such that it ignores Pitch for fixed wings without complaint. And then once another new firmware Stable point release comes out and maybe wait a little bit longer than that change QGC to remove the pitch values for FW and set to 0. Still a bit dangerous since I don't understand how old firmware reacts to pitch=0.

DonLakeFlyer, 2021-02-19:
Firmware change: https://github.com/PX4/PX4-Autopilot/pull/16792

DonLakeFlyer, 2021-02-19:
Maybe @LorenzMeier, @dagar and me on a call to talk through this? It's going to be tricky.

DonLakeFlyer, 2021-02-22:
This has been changed on the Firmware side

closed 2021-02-22 by DonLakeFlyer