mavlink/qgroundcontrol #9444: Stable_V4.1: TCP-MPEG2 Video Stream fails ungracefully -- hangs app

dakejahl, 2021-02-16:
## Expected Behavior
Selecting a video stream with a bad configuration, or when nothing is connected, should causer gstreamer to fail gracefully.

## Current Behavior
When selecting TCP-MPEG2 Video Stream with either a bad configuration or nothing connected, the app will spam error messages to the console and it will hang when you close the app, forcing you to end the process manually (kill -9 or task manager)
 
## Steps to Reproduce:
Just enable TCP-MPEG2 Video Stream option

## System Information
```
VideoReceiverLog: GStreamer error: Could not open resource for reading.
VideoReceiverLog: GStreamer error: GStreamer error: state change failed and some element failed to post a proper error message with the reason for the failure.
VideoReceiverLog: Failed
```

LorenzMeier, 2021-02-16:
We can't be supporting this many streams reasonably. I'm closing this here - please use RTSP / H264 or H265.

closed 2021-02-16 by LorenzMeier

DonLakeFlyer, 2021-02-16:
I"m pretty sure this is Sentera back compat for existing vehicles

DonLakeFlyer, 2021-02-16:
I think it's fine to pull back on formats for the sake of sanity but it likely needs to be announced officially and with some sort of timetable of end of life.