mavlink/qgroundcontrol #9460: dependent 'libs\xz-embedded\linux\lib\xz\xz_crc32.c' does not exist.

mcsauder, 2021-02-19:
This issue just showed up a few days ago for me after a QT update that I mindlessly clicked "OK" for.  There is nothing a web search does to illuminate the subject for me and I'm stuck.  I uninstalled and reinstalled QT on Windows 10 carefully, only selecting installation peripherals and essentials for QT 5.12.10, but the issue persists.

Does anyone have advice for me on this?  Thank you so much in advance.

-Mark

DonLakeFlyer, 2021-02-19:
Update you submodules

closed 2021-02-19 by DonLakeFlyer

mcsauder, 2021-02-19:
Thanks @DonLakeFlyer , but submodules were/are updated and that doesn't fix this issue.  Try out a clean install according to the docs, I've retraced my steps as carefully as I can.  I'll keep working on it and post back if I can fix things.  Thanks again for any advice.

DonLakeFlyer, 2021-02-19:
Delete you entire build directory and build again

mcsauder, 2021-02-19:
I've done that, I can build in Linux QT, it's in windows that's the problem.  Have you tried building in Windows in the past week or so?

mcsauder, 2021-02-19:
This is from a brand new (45GB) installation of QT 5.12.10 in windows 10 with a fresh clone of the QGC repo:
[image]

mcsauder, 2021-02-27:
I've gotten things working again.  For future readers, submodules were not the issue.  QT 5.12.6 is only visible/accessible through the archives and QT 5.12.10 is the next version avilable as a default without selecting the Archives filter.  QT 5.12.10 worked in Windows 10 up until a few weeks ago, but not longer does after a fresh installation, resulting in the error above.  QT 5.12.6 works, after a full uninstall/reinstall. Admittedly, the developer's guide does specifically call out 5.12.6, so this one's on me.  Thanks for your help with the issue!

DonLakeFlyer, 2021-03-02:
Wow, that's a strange one. Although we've seem many problems with using a different minor version of Qt from what is specified. Hence the requirement to use the exact version specified. I've never seen it fail on using a newer patch version.

mcsauder, 2021-04-20:
Hi @DonLakeFlyer ,

I realize this might not be hitting a lot of developer's radar's yet, because for anyone with an established build environment it might not be a problem, but I think this issue needs to be re-opened.  It's back with a vengeance after recent QT updates and regardless of (re)-trying a QT5.12.6 install, a QT5.12.10 install, or a QT6 install, or multiple variations thereof including packages/modules or excluding them, following the instructions in the developers guide simply no longer works.  

Here is the output of installing a full version of 5.12.6, but being selective to match the dev guide produces similar results:

```
01:36:03: Running steps for project qgroundcontrol...
01:36:03: Starting: "C:\Qt\5.12.6\msvc2017_64\bin\qmake.exe" C:\Users\Mark\Documents\GitHub\qgroundcontrol\qgroundcontrol.pro -spec win32-msvc "CONFIG+=debug" "CONFIG+=qml_debug" CONFIG+installer
Project MESSAGE: Qt version 5.12.6
Project MESSAGE: Windows build
Project MESSAGE: Daily Build
Project MESSAGE: QGroundControl Development hide_flight_modes:c45fc9086 2021-04-19 09:08:03 -0600
Project MESSAGE: Debug flavor
Project MESSAGE: Skipping support for Pairing
Project MESSAGE: Using MAVLink dialect 'ardupilotmega'.
Project MESSAGE: Skipping support for Zeroconf (unsupported platform)
Project MESSAGE: Including support for AirMap
Project MESSAGE: Skipping support for Airmap (unsupported platform)
Project MESSAGE: Taisync disabled
Project MESSAGE: Microhard disabled
Project MESSAGE: Skipping support for video streaming (GStreamer libraries not installed)
Project MESSAGE: Installation instructions here: https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoReceiver/README.md
Project MESSAGE: This project is using private headers and will therefore be tied to this specific Qt module build version.
Project MESSAGE: Running this project against other versions of the Qt modules may crash at any arbitrary point.
Project MESSAGE: This is not a bug, but a result of using Qt internals. You have been warned!
WARNING: Failure to find: libs\xz-embedded\linux\lib\xz\xz_crc32.c
WARNING: Failure to find: libs\xz-embedded\linux\lib\xz\xz_crc64.c
WARNING: Failure to find: libs\xz-embedded\linux\lib\xz\xz_dec_lzma2.c
WARNING: Failure to find: libs\xz-embedded\linux\lib\xz\xz_dec_stream.c
WARNING: Failure to find: libs\xz-embedded\linux\include\linux\xz.h
WARNING: Failure to find: libs\xz-embedded\linux\lib\xz\xz_lzma2.h
WARNING: Failure to find: libs\xz-embedded\linux\lib\xz\xz_private.h
WARNING: Failure to find: libs\xz-embedded\linux\lib\xz\xz_stream.h
WARNING: Failure to find: libs\xz-embedded\userspace\xz_config.h
Cannot find file: CONFIG+installer.
01:36:15: The process "C:\Qt\5.12.6\msvc2017_64\bin\qmake.exe" exited with code 2.
Error while building/deploying project qgroundcontrol (kit: Replacement for "Desktop Qt 6.0.3 MinGW 64-bit")
When executing step "qmake"
01:36:15: Elapsed time: 00:12.
```

To reproduce, make sure all current Windows updates have been installed through your system settings, then uninstall QT using the maintenance tool.  Last, follow the Developer's Guide instructions very carefully to install QT 5.12.6 with only the required modules or try again with a full install of 5.12.6, either will produce the same results. demonstrated above.

mcsauder, 2021-04-27:
> Update you submodules

You are absolutely correct.  For anyone using Github Desktop in Windows and expecting submodules to initialize and update automatically when switching between branches, this does not always happen, even though it should: https://github.com/desktop/desktop/pull/5154

As a first recourse, you should install git bash and always run `git submodule update --init --recursive` from the command prompt.  It appears that `--recurse-submodules` in Github Desktop is not always initializing, although it should: https://git-scm.com/book/en/v2/Git-Tools-Submodules