mavlink/qgroundcontrol #9462: Telemetry widget: battery items show depth instead

Williangalvani, 2021-02-19:
## Expected Behavior
The voltage and current should show across multiple QGC restarts
[image]

## Current Behavior
QGC seems to be unable to save the settings(check log), and shows depth instead.
[image]

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior

1. Open QGC and Ardupilot( sub?) SITL
1. Add voltage and current to telemetry bar
1. Kill SITL
1. Click "disconnect" in QGC
2. Close QGC
1. Open QGC

## System Information

QGC master(2ca2cfe) on Arch Linux

Ardupilot SITL (master - https://github.com/ArduPilot/ardupilot/commit/d4aef2dba82c936bd4ff4846f4ffa176072eae8f)

## Log Files and Screenshots

Relevant logs:
```
qrc:/qml/QGroundControl/Controls/QGCFlickableVerticalIndicator.qml:23:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
qrc:/qml/QGroundControl/Controls/QGCFlickableHorizontalIndicator.qml:23:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
qrc:/qml/QGroundControl/FlightDisplay/MultiVehicleList.qml:92: TypeError: Cannot read property 'width' of null
Unknown FactGroup "battery0"
Unknown FactGroup "battery0"
Unknown FactGroup "battery0"
Unknown FactGroup "battery0"
VideoReceiverLog: Stop called on empty URI
Done
After app delete

```

DonLakeFlyer, 2021-02-21:
This is because the battery fact groups are dynamic based on the received telemetry. If you are not connected to a vehicle then you can't get values and the fact groups don't exist. When you connect to a vehicle this will all correct itself and show good values. Making things look better when not connected is a bit tricky. You also cannot setup multiple batteries in the telemetry display unless you are connected. I've tried an failed a simple way to make this better once already. I'll give it a try again at some point.

Williangalvani, 2021-02-22:
When you connect to a vehicle this will all correct itself and show good values.
    
That would be fine, but it's not really happening in this particular case. Maybe it is saving the "depth" instead before the battery facts show up?

DonLakeFlyer, 2021-02-22:
Ok, I'll look.

DonLakeFlyer, 2021-02-22:
Can you paste the contents of your .ini file here? Log output shows the location of the file at boot.

Williangalvani, 2021-02-23:
After clicking "Disconnect", I get the `Unkown FactGroup "battery0"`
And this in the .ini:
```

[TelemetryBarUserSettings-12]
columns\1\rows\1\factGroupName=Vehicle
columns\1\rows\1\factName=AltitudeRelative
columns\1\rows\1\icon=arrow-thick-up.svg
columns\1\rows\1\rangeType=0
columns\1\rows\1\showUnits=true
columns\1\rows\1\text=Alt (Rel)
columns\1\rows\2\factGroupName=Vehicle
columns\1\rows\2\factName=DistanceToHome
columns\1\rows\2\icon=bookmark copy 3.svg
columns\1\rows\2\rangeType=0
columns\1\rows\2\showUnits=true
columns\1\rows\2\text=Distance to Home
columns\1\rows\size=2
columns\2\rows\1\factGroupName=
columns\2\rows\1\factName=
columns\2\rows\1\icon=
columns\2\rows\1\rangeType=0
columns\2\rows\1\showUnits=true
columns\2\rows\1\text=Voltage
columns\2\rows\2\factGroupName=Vehicle
columns\2\rows\2\factName=GroundSpeed
columns\2\rows\2\icon=arrow-simple-right.svg
columns\2\rows\2\rangeType=0
columns\2\rows\2\showUnits=true
columns\2\rows\2\text=Ground Speed
columns\2\rows\size=2
columns\3\rows\1\factGroupName=
columns\3\rows\1\factName=
columns\3\rows\1\icon=
columns\3\rows\1\rangeType=0
columns\3\rows\1\showUnits=false
columns\3\rows\1\text=Current
columns\3\rows\2\factGroupName=Vehicle
columns\3\rows\2\factName=FlightDistance
columns\3\rows\2\icon=travel-walk.svg
columns\3\rows\2\rangeType=0
columns\3\rows\2\showUnits=true
columns\3\rows\2\text=Flight Distance
columns\3\rows\size=2
columns\size=3
fontSize=3
rowCount=2
version=1
```

DonLakeFlyer, 2021-02-23:
Ok, yeah that save is screwed up.

closed 2021-03-12 by DonLakeFlyer