mavlink/qgroundcontrol #9466: Terrain collision detection too sensitive near the Landing Area and Launch (fixed wing patterns)

Antiheavy, 2021-02-20:
The terrain intersection calculation is too sensitive around the Land point (and Launch point too actually) when using typical fixed wing landing and takeoff patterns.  The landing pattern and takeoff line often show as red when they shouldn't.  

Terrain models such as the one QGC is using have all sorts of weird effects at polygon (or post) edges and they use who-knows-what styles of smoothing functions - they just aren't designed for ultra close-up inspection.  The accuracy and fidelity of the terrain model used is not high enough where it is appropriate to be trying to determine a terrain intersection such as a small bump in the ground that is right near where the pilot is standing anyway.

**Suggested improvement:**  ignore terrain intersections within X meters horizontal distance of the Landing and Launch locations.  Might need to play around to determine X, but in past experience I've seen 3-10 meters used.

[image]

DonLakeFlyer, 2021-02-21:
Yeah, I was concerned my last change to this wouldn't really cut it. So no surprise here.

DonLakeFlyer, 2021-03-12:
This specific problem should be fixed now. But leaving this open since it's possibly related to #9495 as well.

DonLakeFlyer, 2021-03-20:
As far as I can tell it's fixed by: https://github.com/mavlink/qgroundcontrol/pull/9547. There will be a new master today sometime you can test with. The Stable RC pointed to here: https://github.com/mavlink/qgroundcontrol/issues/9495 should have the fix in it as well.

DonLakeFlyer, 2021-03-25:
Should be fixed now

closed 2021-03-25 by DonLakeFlyer

Antiheavy, 2021-03-26:
yep, looks like it based on latest daily build testing.

PR #9476 (2021-02-23): Ignore terrain collision on beginning/end of takeoff/land segments
* The start/end of takeoff/land segments are ignored with respect to terrain collision. The distance ignored is 30 meters from beginning/end.
* Fix #9466, #9425