mavlink/qgroundcontrol #9470: qgc v4.1.1 video android app  gstreamer udp h264 not play

lyytest0592, 2021-02-22:
I get the lastest code with v4.1.1,video android app gstreamer udp h264,not play,but rtsp ok,can play.
I had tried the gstreamer demo code with ubuntu 20.0.4.It run ok.
demo code under:
#include <gst/gst.h>
#include<stdio.h>
#include<stdbool.h>

int
main (int argc, char *argv[])
{
  GstElement *pipeline;
  GstBus *bus;
  GstMessage *msg;
  
  bool running    = false;

  /* Initialize GStreamer */
  gst_init (&argc, &argv);

  /* Build the pipeline */
  pipeline =
      gst_parse_launch
      ("playbin uri=udp://192.168.0.155:5600",
      NULL);

  /* Start playing */
  running =gst_element_set_state (pipeline, GST_STATE_PLAYING);
  printf("\r runnig %d",running);
  
  /* Wait until error or EOS */
  bus = gst_element_get_bus (pipeline);
  msg =
      gst_bus_timed_pop_filtered (bus, GST_CLOCK_TIME_NONE,
      GST_MESSAGE_ERROR | GST_MESSAGE_EOS);

  /* Free resources */
  if (msg != NULL)
    gst_message_unref (msg);
  gst_object_unref (bus);
  gst_element_set_state (pipeline, GST_STATE_NULL);
  gst_object_unref (pipeline);
  return 0;
}
the demo run on ubuntu os ok.
and qgc,on GstVideoReceiver.cc
void
GstVideoReceiver::start(const QString& uri, unsigned timeout, int buffer)
{
//    GstBus *bus;
//    GstMessage *msg;

//    if (_needDispatch()) {
//        //enter in
//        QString cachedUri = uri;
//        _slotHandler.dispatch([this, cachedUri, timeout, buffer]() {
//            start(cachedUri, timeout, buffer);
//        });
//        return;
//    }

    if(_pipeline) {
        qCDebug(VideoReceiverLog) << "Already running!" << _uri;
        _dispatchSignal([this](){
            emit onStartComplete(STATUS_INVALID_STATE);
        });
        return;
    }

    if (uri.isEmpty()) {
        qCDebug(VideoReceiverLog) << "Failed because URI is not specified";
        _dispatchSignal([this](){
            emit onStartComplete(STATUS_INVALID_URL);
        });
        return;
    }

    _uri = uri;
    _timeout = timeout;
    _buffer = buffer;

    qDebug()<< "Starting" << _uri << ", buffer" << _buffer;
    qCDebug(VideoReceiverLog) << "Starting" << _uri << ", buffer" << _buffer;

    _endOfStream = false;

    bool running    = false;
    bool pipelineUp = false;

    GstElement* decoderQueue = nullptr;
    GstElement* recorderQueue = nullptr;

    do {
        //add by yuf on 20210220
        //_pipeline = gst_parse_launch("playbin uri=rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov",NULL);
        //running = gst_element_set_state(_pipeline, GST_STATE_PLAYING);
        //rtsp running true
        //_pipeline = gst_parse_launch("playbin uri=udp://192.168.0.135:5600",NULL);
        _pipeline = gst_parse_launch("playbin uri=udp://0.0.0.0:5600",NULL);
        running =gst_element_set_state(_pipeline, GST_STATE_PLAYING);
        //udp running false
//        if(gst_element_set_state(_pipeline, GST_STATE_PLAYING)==GST_STATE_CHANGE_FAILURE)
//        {
//            qDebug()<<"state:GST_STATE_CHANGE_FAILURE";//already print this
//        }
//        if(gst_element_set_state(_pipeline, GST_STATE_PLAYING)==GST_STATE_CHANGE_SUCCESS)
//        {
//            qDebug()<<"state:GST_STATE_CHANGE_SUCCESS";
//        }
//        if(gst_element_set_state(_pipeline, GST_STATE_PLAYING)==GST_STATE_CHANGE_ASYNC)
//        {
//            qDebug()<<"state:GST_STATE_CHANGE_ASYNC";
//        }
//        if(gst_element_set_state(_pipeline, GST_STATE_PLAYING)==GST_STATE_CHANGE_NO_PREROLL)
//        {
//            qDebug()<<"state:GST_STATE_CHANGE_NO_PREROLL";
//        }

        //running = gst_element_set_state(_pipeline, GST_STATE_PLAYING) != GST_STATE_CHANGE_FAILURE;
        //qDebug()<<"running:"<<running;
//        // Wait until error or EOS
//        bus = gst_element_get_bus (_pipeline);
//        msg =gst_bus_timed_pop_filtered (bus, GST_CLOCK_TIME_NONE,
//          (GstMessageType)(GST_MESSAGE_ERROR | GST_MESSAGE_EOS));

//        // Free resources
//        if (msg != NULL)
//          gst_message_unref (msg);
//        gst_object_unref (bus);
//        gst_element_set_state (_pipeline, GST_STATE_NULL);
//        gst_object_unref (_pipeline);
    } while(0);

    if (!running) {
        qCCritical(VideoReceiverLog) << "Failed";

        // In newer versions, the pipeline will clean up all references that are added to it
        if (_pipeline != nullptr) {
            gst_element_set_state(_pipeline, GST_STATE_NULL);
            gst_object_unref(_pipeline);
            _pipeline = nullptr;
        }

        // If we failed before adding items to the pipeline, then clean up
        if (!pipelineUp) {
            if (_recorderValve != nullptr) {
                gst_object_unref(_recorderValve);
                _recorderValve = nullptr;
            }

            if (recorderQueue != nullptr) {
                gst_object_unref(recorderQueue);
                recorderQueue = nullptr;
            }

            if (_decoderValve != nullptr) {
                gst_object_unref(_decoderValve);
                _decoderValve = nullptr;
            }

            if (decoderQueue != nullptr) {
                gst_object_unref(decoderQueue);
                decoderQueue = nullptr;
            }

            if (_tee != nullptr) {
                gst_object_unref(_tee);
                _tee = nullptr;
            }

            if (_source != nullptr) {
                gst_object_unref(_source);
                _source = nullptr;
            }
        }

        _dispatchSignal([this](){
            emit onStartComplete(STATUS_FAIL);
        });
    } else {
        GST_DEBUG_BIN_TO_DOT_FILE(GST_BIN(_pipeline), GST_DEBUG_GRAPH_SHOW_ALL, "pipeline-started");
        qCDebug(VideoReceiverLog) << "Started" << _uri;

        _dispatchSignal([this](){
            emit onStartComplete(STATUS_OK);
        });
    }
}
start function.only write some code,but already running is false,state already is GST_STATE_CHANGE_FAILURE
but rstp running ,the same code expert uri.rstp running is ture.why?
udp not play?would you help me?thanks a lot.

lyytest0592, 2021-02-22:
I already had tried replace udp uri 0.0.0.0:5600,but it already running false.udp not play.and gstreamer had updated 1.18.1,and put the readme telled path.qgroundcontrol/gstreamer-1.0-android-universal-1.18.1

lyytest0592, 2021-02-22:
then I changed the udp port 5601,not 5600,then running is true,but already not to show,still waiting for video

kressinluiz, 2021-07-28:
It seems to be working fine with latest master and Stable_V4.1. I sent a test stream from my PC to qgroundcontrol running on android with:
`
gst-launch-1.0 --gst-debug-1 videotestsrc pattern=ball -v ! x264enc ! rtph264pay ! udpsink host=192.168.2.101 port=5600
`
Although the video record function is not working.

jingwenyi, 2022-03-04:
I have made a test ,use tag v4.0.11 , and gstreamer use  the vision of gstreamer-1.0-devel-x86_64-1.14.4 , android and windows  rtsp can support x-264. Note, gstreamer must use gstreamer-1.0-devel-x86_64-1.14.4.

closed 2024-06-04 by HTRamsey