mavlink/qgroundcontrol #9471: QGC position inaccurate (Ubuntu 20.04/no GPS)

remster, 2021-02-22:
## Expected Behavior
On a desktop/GPS-less 20.04 Ubuntu system, QGC Stable v4.1.1 (as downloaded from [here](https://docs.qgroundcontrol.com/master/en/getting_started/download_and_install.html#ubuntu)) shows inaccurate (and practically useless) position (right is correct).
[image]
On a gps-less Windows, shown QGC position is acurate.

## Current Behavior
Present GCS position only if usefully accurate (~10m) otherwise disregard and show the last map position.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Launch QGroundControl.AppImage

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 20.04
- QGC Version: [e.g. 4.1.1]
- QGC build: stable (downloaded from [here](https://docs.qgroundcontrol.com/master/en/getting_started/download_and_install.html#ubuntu), is that stable? How can i tell if it's stable?)
- Flight Controller: N/A
- Autopilot (with version): N/A

## Detailed Description
Related to https://github.com/mavlink/qgroundcontrol/issues/9085 - where it was suggested the problem can be device specific. I think it is platform (Ubuntu) specific.
geoclue service is up:
```
[remek][~/Downloads]$ sudo systemctl status geoclue.service
● geoclue.service - Location Lookup Service
     Loaded: loaded (/lib/systemd/system/geoclue.service; static; vendor preset: enabled)
     Active: active (running) since Mon 2021-02-22 07:48:43 CET; 6min ago
   Main PID: 34444 (geoclue)
      Tasks: 3 (limit: 18766)
     Memory: 7.6M
     CGroup: /system.slice/geoclue.service
             └─34444 /usr/libexec/geoclue

Feb 22 07:48:43 remek-XPS-2 systemd[1]: Starting Location Lookup Service...
Feb 22 07:48:43 remek-XPS-2 systemd[1]: Started Location Lookup Service.
```
Its `/etc/geoclue/geoclue.conf` is vanilla:
```
# # Desktop ID of application without .desktop part
# [random-app]
#
# # Allowed access to location information?
# allowed=true|false
#
# # Is application a system component?
# system=true|false
#
# # List of UIDs of all users for which this application is allowed location
# # info access, separate by ';'. Keep it empty for allowing it for all users.
# users=

[gnome-datetime-panel]
allowed=true
system=true
users=

[gnome-color-panel]
allowed=true
system=true
users=

[org.gnome.Shell]
allowed=true
system=true
users=

[io.elementary.desktop.agent-geoclue2]
allowed=true
system=true
users=

[epiphany]
allowed=true
system=false
users=

[firefox]
allowed=true
system=false
users=
```
I would add QGC to this list, but can't work out what its "Desktop ID of application without .desktop part" is.

Ubuntu says, no application has requested location services:
[image]

Looking at QGC `::updateLocation` coordinates - it seems as if the lat lng is trimmed to ~100m acurracy.
[image]

It is in my case around 15km off.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

DonLakeFlyer, 2021-02-22:
So if you look here: https://github.com/mavlink/qgroundcontrol/blob/master/deploy/create_linux_appimage.sh#L68. The desktop file is QGroundControl.desktop. So I think the ID is `QGroundControl`. Could you give that a try to see what happens? Otherwise maybe the desktop file generated needs to be updated with some missing thing for ID. I know nothing about desktop files.

remster, 2021-02-23:
Quick experiments yield no results. I understand this is as deep waters for you as they are for me @DonLakeFlyer. I do have a small time budget to investigate this. Please watch this space.

remster, 2021-02-24:
So @DonLakeFlyer - the plot thickens.
I've launched `gnome-maps` and saw exactly the same problem as i saw in QGC - except, QGC didn't show the accuracy radius.
[image]

Both use geoclue that - by default - is configured against `https://location.services.mozilla.com/v1/geolocate?key=geoclue` - except, Firefox itself isn't eating its own dogfood:
[image]
It uses google's - which is why it is correct.

Geoclue can be configured to work against google backend
```
# To use the Google geolocation service instead of mozilla's, simply uncomment
# this url while changing API_KEY to your Google API key and comment out or
# remove the url above.
#
# WARNING: Please make sure that you are complying with the Google's ToS and
#          policies if you uncomment this:
#
# https://developers.google.com/maps/documentation/geolocation/policies
#
# url="https://www.googleapis.com/geolocation/v1/geolocate?key=YOUR_KEY"
```
but i failed:
`Feb 24 07:39:50 remek-XPS-2 geoclue[1807]: Failed to query location: Forbidden` (double-checking earlier that my key is armed).

A PR is born: https://github.com/mavlink/qgroundcontrol/pull/9483

booo, 2022-08-22:
As far as I can see a solution was implemented in https://github.com/mavlink/qgroundcontrol/pull/9483.

closed 2022-08-22 by booo