mavlink/qgroundcontrol #9478: QGC 4.1.1 video on macos is not working

patrickelectric, 2021-02-23:
## Expected Behavior
Video should show

## Current Behavior
Video is not showing

## Steps to Reproduce:
1. Open QGC
2. Video does not show

## System Information
When posting bug reports, include the following information
- Operating System: MacOs 11.2.1
- QGC Version: 4.1.1
- QGC build: stable
- Flight Controller: None
- Autopilot (with version): None

## Log Files and Screenshots
It appears that the problem is that it can't find qmlglsink, from log:
```
 GST_ELEMENT_FACTORY gstelementfactory.c:456:GstElement *gst_element_factory_make(const gchar *, const gchar *): no such element factory "qmlglsink"!
```

patrickelectric, 2021-02-24:
The problem is with the .tar.gz used in deployment by trevis, I was able to fix it remotely, a new version will be provided to fix that.

DonLakeFlyer, 2021-03-16:
@patrickelectric What's going on here?

patrickelectric, 2021-03-16:
some dlls are missing on the gstreamer .tar.gz for macos, I'm planning to put some time and fix that tomorrow or Thursday

DonLakeFlyer, 2021-03-27:
Any updates on this. I'm going to mark a new stable today but this is still and outstanding stable problem.

patrickelectric, 2021-03-27:
There is a dmg with the fix, I'm waiting someone to do the test to me to make sure, but if someone here wants to try:
https://github.com/patrickelectric/qgroundcontrol/releases/tag/test-mac-video-fix

DonLakeFlyer, 2021-03-28:
Ah, ok thanks. You could ask on slck if anyone can test. I won't have time msyelf. I heading out of town for a couple weeks in about 5 minutes.

smnogar, 2021-06-26:
I can confirm that @patrickelectric's fix works on Big Sur. Can we get this fixed in the stable version :)

ES-Alexander, 2021-10-13:
Confirming this works for me and another person, both also on Big Sur 11.6.

> video does not work on daily nor stable 4.1.4

@patemil The fix is only in @patrickelectric's fork at the moment, so no surprises there. He's said he'll try to get back to this in the next week or two (and get it merged if nothing else needs fixing) :-)

mrpollo, 2021-11-30:
@patrickelectric did we ever manage to get this fixed? I can't recall

Vaakefjell, 2022-01-14:
I just had the same issue with the newest version of QGC (4.2.0) - has it not been fixed, or is there something related to my setup?

patrickelectric, 2022-02-24:
Check 4.1.6:
https://github.com/mavlink/qgroundcontrol/releases/tag/v4.1.6

patrickelectric, 2022-02-24:
4.2 will be the next one my list

Vaakefjell, 2022-02-25:
I just tested 4.1.6 and can confirm :)

Vaakefjell, 2022-05-11:
Any progress on adding the fix to the latest version?

ES-Alexander, 2022-05-12:
@Vaakefjell Patrick is away at the moment, and AFAIK hasn't had a chance to work on 4.2 yet.

Not sure if someone from outside Blue Robotics will have looked into it, although given #10140 perhaps the gstreamer version will be changing first. I'm unsure whether that's related to or will indirectly have an impact on the ease of fixing Mac video in `v4.2.x`.

PR #10291 (2022-05-13): Mac video fix v4.2
Fix #9478
closed 2022-05-13 by patrickelectric