mavlink/qgroundcontrol #9482: Unable to establish connection with PX4 both through USB and telemetry

ShigiDono, 2021-02-24:
## Expected Behavior

Usually, UAV connects in few seconds to QGC and I am able to change the state, etc.

## Current Behavior

It became impossible to send any sort of commands to PX4.

## Steps to Reproduce:

1. Upload >1.11.3 firmware (tried with the master firmware also)
2. Try to connect with QGC >4.1

## System Information

- Operating System: macOS Catalina 10.15.7
- QGC Version: >4.1
- QGC build: daily, 4.1.1
- Flight Controller: Holybro PX4 (fmuv5)
- Autopilot (with version): 1.11.3, master

## Detailed Description

Firstly, there are a bunch of connection errors showing at first:
```
Mission transfer failed. Error: Mission request list failed, maximum retries exceeded.
GeoFence transfer failed. Error: Mission request list failed, maximum retries exceeded.
Rally Point transfer failed. Error: Mission request list failed, maximum retries exceeded.
```

Also, If I am to send ARM/DISARM/etc command with my gamepad it returns:
```
Vehicle did not respond to command: MAV_CMD_COMPONENT_ARM_DISARM
```

It looks like there is some sort of problem with mavlink communication.

At first, I was using custom firmware and custom QGC build, but the same errors continue appearing even after switching to 1.11.3 and the daily build of QGC.
Though, I'm still able to control everything with 4.0.11 even while using the custom firmware.
A similar issue happened on Linux machine.

## Log Files and Screenshots

[image]

[qgc_log.txt](https://github.com/mavlink/qgroundcontrol/files/6034402/qgc_log.txt)

steveb2014, 2022-03-03:
Hi ShigiDono -  Did you ever figure this out?

SMRazaRizvi96, 2023-08-18:
Did you find a solution?

ansonyuu, 2025-01-17:
Also wondering!

ShigiDono, 2025-01-17:
Solved by updating both qground control and px4 firmware. Usually nightly builds work fine, if not -- you can build from source.

closed 2025-01-17 by ShigiDono

ansonyuu, 2025-01-17:
> ID: <mavlink/qgroundcontrol/issues/9482/2597448182 @ github. com>

Thanks so much for the speedy reply! Appreciate it.

I'm wondering which is the best link and how to upload the build (if at all) to QGC. We're beginners and trying to figure it out.

We are also running into something where the software does not identify any gyros, despite using pixhawk4

Thanks so much again.

Kindly,

Anson