mavlink/qgroundcontrol #9484: Joystick: MANUAL_CONTROL messages are not regular

Williangalvani, 2021-02-24:
## Expected Behavior
MANUAL_CONTROL messages should be sent with a regular interval when there is a joystick connected (or virtual joystick is enabled.

## Current Behavior
Every 30 seconds, something disrupts the messages, causing large latency spikes

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Open SITL (any?)
2. Open QGC
3. Close QGC
4. Check latest telemetry log

## System Information

- Operating System: Arch Linux
- QGC build: QGC 88b2b8f5
- Ardusub SITL master

## Detailed Description
This causes ArduSub to complain about "Lost Manual Control". Which is correct. While this is usually pretty quick, I just looked a a users tlog where this freeze peaked at 25 seconds! Every time the blue line rises above the orange one, ArduSub emits a "Manual Control Lost" warning.

[image]
User log:
[2021-02-17 17-04-38.zip](https://github.com/mavlink/qgroundcontrol/files/6036717/2021-02-17.17-04-38.zip)

## Investigation

I did some digging already, and this seems to be caused by [QGCMapEngine::testInternet()](https://github.com/mavlink/qgroundcontrol/blob/master/src/QtLocationPlugin/QGCMapEngine.cpp#L390) (the only relevant timer fired every 30s). 

This is a regular log of SITL:
[image]

This is SITL with `checkInternet=false` in `[General]` (is there no UI for it?)
 
[image]

So it seems that on some setups, the internet check can take long enough to disrupt regular operation of QGC...

DonLakeFlyer, 2021-02-25:
> So it seems that on some setups, the internet check can take long enough to disrupt regular operation of QGC...

But the question is why? That should be happening on its own thread.

jafrado, 2021-03-07:
Hmm. Interesting. My suggestion would be to disable roaming and scanning over WiFi when flying. The issue is that most modern IP stacks (Win/Linux/iOS) will temporarily go off-channel when scanning for new networks (if you understand how 802.11 scanning works you will get it). The net effect is that latency will continue to stack up and as traffic increases, delays may become unbounded and/or create loss. I had this problem while developing some commercial real-time wireless streaming products. One of the classical ones was "gaps in wireless streaming" - the fix - disable scanning while streaming. I think if you google "scanning" one would find some good reads on this topic today.

DonLakeFlyer, 2021-03-10:
Wow, nasty.

> My suggestion would be to disable roaming and scanning over WiFi when flying.

That should be doable since QGC knows if it has any vehicle connected and also has a flying state.

jafrado, 2021-03-10:
yep, very nasty. We don't want to be off-channel while sending or receiving mavlink. Yes, lots of options to fix, here's a good read on the topic: https://blogs.gnome.org/dcbw/2016/05/16/networkmanager-and-wifi-scans/

patrickelectric, 2021-04-15:
The main problem with the current implementation is that the QGCCacheWorker
does not have an event loop (not calling exec).
Since network function that are not async (such as `waitConnection`) needs an eventloop,
the Fuctor will run in the main thread.

To fix this behaviour, the implementation was moved from to an async approach with signals.
Check #9591