mavlink/qgroundcontrol #9486: "Communication regained" modal popup a bit of a nuisance

remster, 2021-02-25:
## Expected Behavior
Communication lost or regained **arguably** aren't the kind of modal dialogs that the **pilot** needs to be forced to dismiss when flying the drone. 

## Current Behavior
When the drone looses the connection QGC clearly shows that in red in the top right corner. When the connection is regained, it turns green clearly showing the current status. Bit then for good measure it confronts the pilot (already well informed) with a dialog that it needs to dismiss.
[image]

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Launch simulator (e.g.: `make px4_sitl_default jmavsim`)
2. Launch QGC and wait for it to snap to the drone 
3. Kill the simulator and wait for QGC to show that in the top right corner
4 Launch the simulator again
5 "Connection regained" dialog pops

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 20.04
- QGC Version: 4.1.1
- QGC build: stable
- Flight Controller: not sure
- Autopilot (with version): N/A

DonLakeFlyer, 2021-02-25:
If you happen to be on the Fly View when you lose commands you are correct. But if you are in any other view you do not get the toolbar indicator of comm loss. This is why the popup is there as well. Comm lost is a pretty serious problem. So it's as much in the operators face as possible.

pklapperich, 2021-02-26:
I think the communication lots popup is OK. But I would like to see the modal automatically dismiss when communication is regained. Maybe the dialog for "communication lost" could have a checkbox for "Automatically close" like some dialogs in other apps have check boxes like "don't show again"?

Alternatively, the Fly View displays "Communication Lost ... [Disconnect]" in the top bar. Why not do the same thing on the Plan View, Vehicle Setup View, and Application Settings view?

remster, 2021-02-26:
> But if you are in any other view you do not get the toolbar indicator of comm loss.
My annoyance was with the FlyView and during an actual flight. Maybe this creates a precedent where each dialog would present itself differently depending on the view. I can see this would be a big change and necessitate UX debates. Please feel free to close this issue if you indeed think there is nothing to address.

DonLakeFlyer, 2021-02-26:
> Please feel free to close this issue if you indeed think there is nothing to address.

Previous to 4.1 both incoming status_text messages from the vehicle as well a most QGC originated errors/messages showed up in the orange modeless message area popup thing. This caused all sort of confusion are to what came from where. In 4.1 the two types of messages were separated and only incoming vehicle status shows in the orange modeless area now. QGC messages now show in a popup dialog. So although the amount of "messaging" is exactly the same the way they are shown now greatly increases the annoyance factor. So that said I do think something should change here. It may require something simple/save for stable and then something more complex (like making it always show in the toolbar) for master going forward.

DonLakeFlyer, 2021-03-20:
So I looked at this and decided to remove the modal popups for comm lost/regained. I left the modal pop up for when the primary link changes due to comm loss. In looking back at how previous QGC worked this isn't much less unsafe than before. The problem is that the toolbar only shows comm lost indicator when on the Fly View. If you are on a different view you still get the audio indication though.

closed 2021-03-25 by DonLakeFlyer