mavlink/qgroundcontrol #9490: Communication Lost / Regained logic is broken

dakejahl, 2021-02-25:
On branch Stable_V4.1

## Current Behavior
When unplugging a radio from the GCS, or unplugging the direct USB connection from pixhawk to GCS, QGC properly detects the comms lost event, updates the UI, and notifies the user. 

However when the connection is re-established, QGC does not detect the the comms regained event. The "Disconnect" button remains on the top toolbar and no "communication regained" message is announced. There is some weird partial update to the UI, the name of the link appears to the right of the battery indicator after reconnecting
[image]

## Steps to Reproduce:
1. Unplug / replug your pixhawk or radio

dakejahl, 2021-02-25:
I just tested against simulation and strangely it works with that. Maybe something to do with sik radios?

DonLakeFlyer, 2021-02-26:
Depending on the OS the USB can come back on a different port. This makes QGC think you have multiple links connected hence the toolbar indicator. If you click it you should see the multiple links dropdown. That said the comm list indication in the toolbar should go away since it does have good comms on at least one link. And then if you click the link the dropdown should tell you that comm was lost only on the previous link whereas the current link has a good connection.

DonLakeFlyer, 2021-02-26:
It should be fixed but given the fact that this will only happen on USB connection it's not critical for stable. I'll take a look anyway to see if it indicates a worse problem that may affect stable.

DonLakeFlyer, 2021-11-08:
So in the case where the same port ends up being reused when you re-plug there is no great way to detect that. If you disconnect autoconnect will reconnect. The reason is that QGC does not track a hardware connection being pulled. This is on purpose. The reason being that when the last link goes away the whole Vehicle goes away and you lose all vehicle state in the ui. The thinking is that it's better on hardware link loss to continue to show vehicle state to the user so they can see say where the vehicle was last.

So anyway given the way things work now there isn't a better way to handle this. Would require a bunch of work to track serial ports in a zombie disconnected state and try to reconnect. Doable I think but disn't seem worth the effort.

closed 2021-11-08 by DonLakeFlyer