mavlink/qgroundcontrol #9495: terrain checker fails pretty often on large missions

Antiheavy, 2021-03-01:
When downloading a large(ish) mission from an autopilot (e.g. upon connection) the terrain checker quite often fails on some number of the waypoints.  

Repo:
1) upload a medium large(ish) mission (100's of waypoints) to an autopilot
2) "download" the mission
3) observe the terrain checker fail part way through

Below is tested with 4.1.x stable on Windows connected to an autopilot over USB:

[image]

DonLakeFlyer, 2021-03-02:
Can you attach that plan?

Antiheavy, 2021-03-02:
[terrain-check-fails-after-round-trip.zip](https://github.com/mavlink/qgroundcontrol/files/6072174/terrain-check-fails-after-round-trip.zip)

attached here

DonLakeFlyer, 2021-03-18:
Repro:
* Boot QGC
* Load big plan on vehicle
* Shutdown vehicle and disconnect
* Reconnect vehicle
* Boom

Antiheavy, 2021-03-18:
> Boom

yeah, but I find the terrain checker "gives up" at different spots each time.

DonLakeFlyer, 2021-03-18:
There might be multiple paths. It's a signaling problem with home altitude value updates not working it's way through all the items. You can see how in the terrain profile the item indicators are in the wrong place.

DonLakeFlyer, 2021-03-18:
There may be a separate bug but this one is caused by the Vehicle updating it home position altitude multiples times after connection.

DonLakeFlyer, 2021-03-18:
> different spots each time.

It all depends on the timing of the vehicle updating the altitude of the home position.

DonLakeFlyer, 2021-03-19:
@Antiheavy Can you try this new Stable RC https://qgroundcontrol.s3-us-west-2.amazonaws.com/builds/Stable_V4.1/QGroundControl-installer.exe? I'm not sure if there is one bug here or multiple. I know I fixed at least one.

M-Skelton, 2021-03-19:
@DonLakeFlyer @Antiheavy 

I test this a bunch today. Everything works as expected. 

Setup:

- PX4 Vanilla SITL
- QGC build from above comment
- Large surveys (300+ WPs) w/ terrain following enabled & tolerance set to 0ft. 

Here were the two procedures I tested:

Procedure 1:
1. Plan mission w/ large survey (300, 700, 1000 WPs)
2. Upload to vehicle
3. Disconnect from vehicle (killed SITL)
4. Restart SITL and connect to QGC
5. Check Plan screen, Fly screen, & mission checker for errors

Procedure 2:
1. Plan mission w/ large survey (300, 700, 1000 WPs)
2. Upload to vehicle
3. Download from vehicle
4. Check mission checker for errors
5. Check Fly screen for errors
5. Upload again 
6. Download again 
7. Check mission checker for errors
8. Check Fly screen for errors

Antiheavy, 2021-03-20:
@DonLakeFlyer  and @M-Skelton 

This seems to be working better now.  I tested this a bunch today with my set-up (autopilot w/GPS connected to PC over USB).  I have yet to be able to get the terrain checker to fail when downloading or re-connecting to the autopilot whereas before it was pretty repeatable. 

One thing I noticed is the mission "double paints" on the screen, which isn't too annoying, just wondering if it is expected (or if it is related to https://github.com/mavlink/qgroundcontrol/issues/9425)
[image]

DonLakeFlyer, 2021-03-20:
> One thing I noticed is the mission "double paints" on the screen, which isn't too annoying

Yes there is a bunch of over-signalling going on here which causes multiple updates to happen. It's too dangerous to fix in Stable. I can take a look on master. Can you add a new Issue for this with some repro steps.

closed 2021-03-20 by DonLakeFlyer