mavlink/qgroundcontrol #9499: Embed mission items into .plan file for Structure Scan

julianoes, 2021-03-02:
I am writing on .plan import functionality for MAVSDK and was looking through the various .plan files which can be exported from QGC.

I found that normal waypoint missions, as well as surveys populate the list of mission items which allows for easy exporting to MAVLink mission items. However, I realized that the Structure Scan .plan does not save the actual mission items but only the locations, so more interpretation/conversation is necessary to get to the mission items to send to an autopilot.

Is there a specific reason for this difference? Is this where the .plan format is headed in general and relying on the MAVLink mission items is wrong anyway?
Or could the actual mission items be added in the future thus enabling the Structure Scan .plan to be imported in other places?

@DonLakeFlyer I would appreciate your response, thank you.

DonLakeFlyer, 2021-03-02:
The list of actual mission items for a Pattern was something that had been added over time, for two reasons:
* If the internal pattern generation code changes in a newer QGC you can load and old plan with a newer and the individual items from the plan files are used such that you end up with exactly the same mission.
* To allow other GCS to upload plans without needing to understand how to generate a pattern.

I just never got around to adding that same support to Structure Scan.

julianoes, 2021-03-03:
Got it. At this point it would be a nice to have, until then MAVSDK can't import Structure Scans and will return "Unsupported".

closed 2026-10-04 by github-actions[bot]