mavlink/qgroundcontrol #9501: Huge performance loss when a lot of "picture taken" icons are displayed

Davidsastresas, 2021-03-02:
When running a mission, if mavlink messages announcing pictures taken from a survey arrive we draw a camera icon over the map. For a few icons QGC works great, but on large surveys ( lots of icons displayed ) QGC starts to behave badly, super laggy. It is 100% connected to the number of camera icons in the map. It seems that has a huge impact in screen rendering or QML engine has a hard time to display them all. 

I took a quick look at it and we are not doing anything different from any other map icons, so maybe it is just the mapquickitem being not super optimized for large amounts of icons? I suspect this comes from qt/qml performance of this qml type directly. Does anybody know anything about this? I would love to help, I thought on limiting the number of picture taken icons, similar to how we limit the lenght of the vehicle trajectory, but maybe there is actually a solution preserving the icons?

Thanks!

Antiheavy, 2021-03-02:
We've noticed this in the past and had to stop using the photo icons altogether.   It's a real bummer because most users expect some sort of photo location feedback when flying mapping missions.  Of course, there can be 1000's of photos taken on some mapping missions so whatever the solution ends up being, it needs to be performant/efficient.

Davidsastresas, 2021-03-02:
Thank you, lets see if any of the devs more familiarized with the inners of qml, etc sees the issue and points me to the right direction for working over it. Thanks!

DonLakeFlyer, 2021-03-02:
I know about it but haven't looked at it to figure it out. 

@Davidsastresas I wonder if it is related to using too many display of the camera icon which I think is an svg file. Could you try changing the camera indicator to a regular qml circular something using a Rectangle and see if the perf is still bad?

Davidsastresas, 2021-03-02:
@DonLakeFlyer  Yes, it is a .svg file. Sure, I will test it this week. Besides testing simple qml shapes, do you think it is worth testing other icon format, or any other way of preserving the icon? Thanks!

DonLakeFlyer, 2021-03-02:
> do you think it is worth testing other icon format, or any other way of preserving the icon?

Not sure yet. Theoretically the best possible perf should be from using OpenGL drawing and then you instance the image with a shader such that it is not dup'ed for each icon. But that a bit tricky if you don't know OpenGL.

Davidsastresas, 2021-03-04:
I tested with regular Qml types and there is no performance issue. I made a camera like icon based on Qml rectangles, the PR is here: #9513

I checked the openGL shader stuff and I agree that would be super nice, but I don't have the time right now to properly study that. At some point I will. In the meantime, the above PR solves the performance issue and still looks somehow like a camera.

booo, 2021-10-10:
As far as I can see this bug was resolved with https://github.com/mavlink/qgroundcontrol/pull/9513 Please reopen if you think there is something left to do.

If we want to go the openGL shader route I suggest we create another ticket for that.

closed 2021-10-10 by booo