mavlink/qgroundcontrol #9508: Video not showing in 4.1.1

katzfey, 2021-03-03:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
I expect to see the video display from my drone camera when using QGC. This works fine in earlier versions but stopped working in 4.1.1.

## Current Behavior
The RTSP connection appears to start normally but no video appears.

## Steps to Reproduce:
1. Using Ubuntu 16.04 (gstreamer version 1.8.3), clone and build gst-rtsp-server (https://github.com/GStreamer/gst-rtsp-server) using branch 1.8.3
2. In the examples directory run ./test-launch "( videotestsrc ! x264enc ! rtph264pay name=pay0 pt=96 )"
3. Open QGC and setup the RTSP parameters to point to your server
4. There will be no video shown
5. Try the same thing with QGC 4.0.11 and you will see the video

## System Information
- Operating System: macOS High Sierra 10.13.6
- QGC Version: 4.1.1
- QGC build: stable
- Flight Controller: ModalAI voxl. (But also recreated using an Ubuntu 16.04 box as noted above)
- Autopilot (with version): Currently testing without an autopilot attached.

## Detailed Description
I looked at the GStreamer debug output at both the server side and the client side by using GST_DEBUG=3,rtsp*:7. I looks like the connection is fine on both 4.0.11 and 4.1.1. Tcpdump shows frames are flowing.

## Log Files and Screenshots
N/A

xdwgood, 2021-03-09:
I have the same problem.

patrickelectric, 2021-03-09:
If you are using ubuntu, please check the latest LTS version (Ubuntu 20.04 LTS).

katzfey, 2021-03-13:
Okay, yes, we tested with Ubuntu 20.04 and it does work. So I guess Ubuntu 18.04 isn't really supported then on 4.1.1.

patrickelectric, 2021-03-18:
Duplicated of: #9478

closed 2021-03-18 by patrickelectric

asmirnou, 2021-03-18:
@patrickelectric FYI, the video doesn't work on Ubuntu 18.04 too, not only on Mac OS.

patrickelectric, 2021-03-18:
> @patrickelectric FYI, the video doesn't work on Ubuntu 18.04 too, not only on Mac OS.

Please use the latest ubuntu lts version, 20.04