mavlink/qgroundcontrol #9514: Fixed Wing Landing loiter flips when dragging (bug)

Antiheavy, 2021-03-04:
## Expected Behavior
Landing loiter should stay showing CW or CCW when dragging Land or Loiter point.

## Current Behavior
Landing Loiter flips direction with dragging Land or Loiter points.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Plan a fixed wing landing pattern
2. uncheck "Loiter clockwise", observe loiter direction flip sides
3. drag Land waypoint, observe loiter direction flip
4. drag Loiter point, observe loiter direction flip again
5. repeat.
6. dance.

## System Information
When posting bug reports, include the following information
- Operating System: Windows
- QGC Version: 4.1.x stable
- QGC build: stable from appveyor

## Log Files and Screenshots
[image]

DonLakeFlyer, 2021-03-25:
Should be fixed now

closed 2021-03-25 by DonLakeFlyer

Antiheavy, 2021-03-26:
yep, looks like it based on latest daily build testing.  no more flipping.

PR #9550 (2021-03-23): Plan: Fix Loiter directions
* One form of recalc code wasn't taking into account loiter direction setting
* Fix #9514