mavlink/qgroundcontrol #9537: Android File Saving/Opening

patricktnelson, 2021-03-16:
Installing QGC stable 4.1.1 on Android 10 won't allow proper file saving/opening if 4.1.1 is the first version of QGC to be installed on device. 4.1.1 doesn’t request permission to access files, only photos and media. A workaround for this is installing 4.0, which does work correctly, with file explorer opening on right, and then update to 4.1.1. This carries over the file saving permissions.

see:
https://discuss.px4.io/t/qgroundcontrol-android-file-sharing/9741/3

LorenzMeier, 2021-03-17:
Thanks for reporting!

DonLakeFlyer, 2021-03-20:
@patrickelectric Is your Documents location on an external SD card? Or is it using internal storage?

patricktnelson, 2021-03-20:
My Documents location is on internal storage. I tried the whole, "create the correct QGC file save path" to fix problem, but it said I didn't have permission to save there.

DonLakeFlyer, 2021-03-20:
Ok, I have one thing to try. I'm worried that this won't be fixable until I upgrade to a newer Qt which has better Android 10 support.

DonLakeFlyer, 2021-03-20:
Can you try this apk: https://qgroundcontrol.s3-us-west-2.amazonaws.com/DonLakeFlyer/android-build-debug.apk. This is a stable build with some tweaks to try a fix.

patricktnelson, 2021-03-20:
That did it. It correctly prompted me to obtain permission to access files. The file access to the right side didn't work the first time I ran QGC, but then I force stopped and relaunched. Then it worked as it should. I successfully saved and loaded missions.

closed 2021-07-02 by DonLakeFlyer