mavlink/qgroundcontrol #9546: Does not create Altitude type "Terrain Frame" in Surveys Plans

framosnt, 2021-03-19:
Does not create the Altitude type "Terrain Structure" in Plans

Situation:
Use the QGROUND for drones with Radars or Lidar for RangeFinder for the drone to maintain the altitude of the distance sensor (follow terrain), so that it makes a Survey flight plan with the Terrain Frame option, which is determined in the START MISSION, it is also accepted in the creation of individual Waypoints, but, it is not created when used in the Survey option with "Follow Terrain", that for my flight tests, it would be enough to create the waypoints with the Option "MAV_FRAME_GLOBAL_TERRAIN_ALT", that's it!
----

## Expected Behavior
Please tell us what should happen as detailed as possible.

## Current Behavior
Please describe what happens instead of the expected behavior.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1.
2.
3.
4.

## System Information
When posting bug reports, include the following information
- Operating System: [e.g. iOS 12.1, Windows 8, macOS Mojave 10.14.1, Ubuntu 14]
- QGC Version: [e.g. 3.4.4]
- QGC build: [e.g., daily, stable, self-built from source, etc...]
- Flight Controller: [e.g., Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.]
- Autopilot (with version): [e.g., PX4 1.8.1, Ardupilot Copter 3.6.1]

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

DonLakeFlyer, 2021-03-20:
@LorenzMeier What's the status of `MAV_FRAME_GLOBAL_TERRAIN_ALT` in PX4? Two questions:
* Does it work with Lidar vehicle?
* Does it support the mavlink terrain tile protocol?

DonLakeFlyer, 2021-03-20:
@framosnt The "Follow Terrain" option in Patterns is not meant to work with a mavlink terrain frame. It just calculates absolute waypoint heights based on terrain data from the map. That said QGC does support `MAV_FRAME_GLOBAL_TERRAIN_ALT` but is it currently turned off for PX4 because I didn't think PX4 supported it yet. Waiting on answer that that and then I'll see if there is something I need to update to turn it on.

framosnt, 2021-03-22:
> The "Follow Terrain" option in Patterns is not meant to work with a mavlink terrain frame. It just calculates absolute waypoint heights based on terrain data from the map. That said QGC does support [...]

Hi @DonLakeFlyer,

I am use Ardupilot only, can you turn it on only to Ardupilot ?  I need this function activated, or I will have to make a Customized version, can you refer me to someone who has the knowledge to make Customizations for me ??? I even tried, but, I'm rusty to do it myself, I need help, even if I have to pay for the service.

DonLakeFlyer, 2021-03-24:
> can you turn it on only to Ardupilot

In Mission Start if you are not connected to a Vehicle make sure your Vehicle Info/Firmware Type is ArduPilot. In the All Altitudes drop down select either Terrain Frame or Mixed Modes depending on what you want to use.

DonLakeFlyer, 2021-03-24:
Terrain frame should be turned on for PX4 it's coming soon.

framosnt, 2021-03-24:
> > can you turn it on only to Ardupilot
> 
> In Mission Start if you are not connected to a Vehicle make sure your Vehicle Info/Firmware Type is ArduPilot. In the All Altitudes drop down select either Terrain Frame or Mixed Modes depending on what you want to use.

This process is correct only to waypoints, when you choice SURVEY, all waypoints created after upload this plan will be not “terrain frame”, will be “absolut alt”, and this method isn’t agree with radar or rangefinder “down” / Terrain following method. 

After upload plan, this can be review using “download” plan, will be real plan and waypoints configuration, altitude was “Absolut Global Alt”. 

I need this Survey generator create waypoints settings to “MAV Frame Global Terrain Alt” !

DonLakeFlyer, 2021-03-25:
Ah, ok I see now.

DonLakeFlyer, 2021-03-27:
> This process is correct only to waypoints, when you choice SURVEY, all waypoints created after upload this plan will be not “terrain frame”, will be “absolut alt”,

Ah, ok. Understand now. You are correct this isn't supported yet in Patterns.

framosnt, 2021-03-27:
> > This process is correct only to waypoints, when you choice SURVEY, all waypoints created after upload this plan will be not “terrain frame”, will be “absolut alt”,
> 
> Ah, ok. Understand now. You are correct this isn't supported yet in Patterns.

I need to activate this function, or I will have to make a customized version, can you tell me someone who has the knowledge to make customizations for me?
send contact to my email: framosnt@gmail.com

buckker, 2021-04-18:
@DonLakeFlyer After switching to QGC I'm facing the same problem like framosnt. I also did a post here:

https://discuss.ardupilot.org/t/how-to-create-a-rangefinder-based-terrain-following-mission-with-qgc/69895

Will the feature be available a little bit earlier in a daily build or do we have to wait until the 4.2 is released?

DonLakeFlyer, 2021-07-07:
This is available in daily builds now

closed 2021-07-07 by DonLakeFlyer