mavlink/qgroundcontrol #9551: Problems compiling [RELEASE] versions of recent (~4.1) custom-example per instructions

rworrall, 2021-03-22:
Building recent (including 3/21/21) daily / 4.1 source trees of QGC.  Following instructions for using the custom-example allows debug builds but release builds fail with compile error.
----

## Expected Behavior
Successful build of debug and release after using custom-example

## Current Behavior
Successful DEBUG build, but failed RELEASE build with compiler error:
"

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. git clone https://github.com/mavlink/qgroundcontrol.git --recursive
2. cd qgroundcontrol
3. mv custom-example custom
4. cd custom
5. qgroundcontrol/custom$ ./updateqrc.py 
6. Select RELEASE profile & build
7. Observe error: "No resource path for file: C:\Users\rtw\daily\qgroundcontrol\custom\res\CustomFlyViewOverlay.qml"

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10/64 MSVC2007
- QGC Version: 4.1.x
- QGC build: daily, stable, and self-built from source
- Flight Controller: N/A
- Autopilot (with version): N/A

## Detailed Description

Seems to be related to custom.qrc as below:

 28  <file alias="Custom/Widgets/CustomVehicleButton.qml">res/Custom/Widgets/CustomVehicleButton.qml</file>
 29      <file alias="Custom/Widgets/qmldir">res/Custom/Widgets/qmldir</file>
 30     </qresource>
 31     <qresource prefix="/qml">
 32    <file alias="QGroundControl/FlightDisplay/FlyViewCustomLayer.qml">res/CustomFlyViewOverlay.qml</file>
 33     </qresource>
 34 </RCC>

---------

Moving line (32) above to directly follow line (29) seems to allow RELEASE builds to be created.

DonLakeFlyer, 2021-03-25:
This one is totally bizarre. All I can guess is there is some bug in the qml compiler. Trying to find a work-around.

DonLakeFlyer, 2021-03-25:
Nope. The exclusions were wrong. Strange that it even worked at all for debug builds.

rworrall, 2021-05-14:
Agree.  Now working, thank you.

closed 2021-05-14 by rworrall