mavlink/qgroundcontrol #9556: Terrain following is dependent on order of application of survey settings

M-Skelton, 2021-03-24:
## Expected Behavior
Terrain following workflow should not be dependent on application order

## Current Behavior
Use of terrain data depends on when the feature is enabled during survey planning. If the feature is enabled before a preset is applied the terrain data is not applied (or used?) correctly. 

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Plan/Upload a mission
2. Enable terrain following
3. Select/apply a preset
4. Upload mission to vehicle
5. Check mission on Fly screen - TF data will not be included in mission plan
6. Return to Plan screen 
7. Toggle terrain following off then on again
8. Upload mission to vehicle
9. Check Fly screen - terrain data is included

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10
- QGC Version: See attached screenshot (4.2.?? maybe?)
- QGC build: Daily
[image]

- Flight Controller: SITL (ubuntu VM host in virtual box)
- Autopilot (with version): PX4 v1.11.0rc

## Detailed Description
Maybe this is the expected behavior? Seems weird to me to have the settings laid out the way they are if they are dependent on order of application. 

This could be harmful if someone thought they enabled terrain following to avoid something like rolling hills for a low altitude flight but the feature was not applied correctly because they turned terrain following on then applied a survey preset. This potentially leads to unexpected terrain interaction for the user. 

## Log Files and Screenshots
Video:

https://user-images.githubusercontent.com/28786472/112347702-16795c00-8c95-11eb-8d96-8c4a2b902dd2.mp4

DonLakeFlyer, 2021-03-24:
Applying a preset is going to override anything you did before that based on the what is in the Preset. Not sure what else would be expected?

M-Skelton, 2021-03-24:
It's confusing to me because if a preset is made with terrain following enabled, when the preset is applied the terrain following box does not update like the overlap and camera boxes do. 

Video:

https://user-images.githubusercontent.com/28786472/112359357-30b83780-8c9f-11eb-913b-db814d571fd5.mp4

DonLakeFlyer, 2021-03-24:
Ah, ok. Yup, signaling bug. Thanks.

PR #9560 (2021-03-27): Plan: Fix Follow terrain preset
Fix #9556
closed 2021-03-27 by DonLakeFlyer