mavlink/qgroundcontrol #9576: 2D map customization

cnag-steven, 2021-04-05:
Hi,
Is it possible to integrate the 2D customized files (jpeg,tif.png  image formats)in the existing  QGC map?
If possible where to start with? any suggestions would be really helpful.

Thanks in advance!!!

DonLakeFlyer, 2021-04-05:
Not possible. Although I think you can use mapbox to created your own hosted maps and then hook up that way.

closed 2021-04-05 by DonLakeFlyer

cnag-steven, 2021-04-07:
> Not possible. Although I think you can use mapbox to created your own hosted maps and then hook up that way.

Hi @DonLakeFlyer ,
Thanks for the reply. should i add external files(jpeg,png ,tif) in mapbox API and then call it in QGC? Is that how you are suggesting to integrate the 2D files(image files) in the existing map?
please tell me how should i add external files in mapbox API .
Thank You