mavlink/qgroundcontrol #9578: viewing multiple debug values in Mavlink Inspector

Assaf-Friedman, 2021-04-05:
Hello,

in qgroundcontrol v3.5.1 I am able to graph multiple debug values in mavlink inspector , however with the same working code, if I use qgroundcontrol v4.1.2 I can not graph multiple values.
I can find the values under Named_Value_float, but they alternate under the same name and can not be separated. 
could not find any reference here or elsewhere to solve this issue. 

I'm using the qgroundcontrol on a windows machine, pixhawk 1 and pixhawk 4 controllers runing px4 v1.9.4

DonLakeFlyer, 2021-04-05:
Dup of #8454

Assaf-Friedman, 2021-04-07:
Hello,
**this is not the same issue as  #8454 to my understanding**.
I am trying to view single value in a graph, however if I publish more then 1 value as debug  value as explained here: https://docs.px4.io/master/en/debug/debug_values.html
it is not possible, since they all display in 1 field and the name keep changing.
so it practically limit me to use only single debug value at a time per compiled firmware and this is not very practical since some times I need more. the values are single value and does not have to be on the same graph.

DonLakeFlyer, 2021-04-10:
Ah, ok right. This is different.

joseacain, 2021-09-16:
just ran into this same issue as well. is there an alternate way to view the named debug value floats like outputting to a log file? for some reason, when i try and change SDLOG_MODE, it won't let me go higher than 15 and debug is bit 5 in that bitmask.