mavlink/qgroundcontrol #9579: No Sound Ubuntu 20.04

mrawding, 2021-04-05:
Issue: 
Both the latest .AppImage of QGroundcontrol and building from source with QT 5.12.6 does not support sound on Ubuntu 20. 

Steps to reproduce:

1) Download Qgroundcontrol app
2) remove modemanager
3) install gstreamer dependencies and speech-dispatcher dependencies as described
4) Fix all permissions as described in instructions.
5) run Qgroundcontrol

Result= no sound.

The sound is a very important safety feature of groundcontrol. It is not working for any version of Qgroundcontrol. I have verified that speech dispatcher is working seperately with spd-say. No matter which version I try, new or old, qgroundcontrol will not produce sound. No errors are thrown for what I can tell.

DonLakeFlyer, 2021-07-02:
This is working fine for me on 20.04. My suggestion would be to ask around on the QGC clack channel to see if anyone else is seeing the problem. If so I can dig some more and reopen.

closed 2021-07-02 by DonLakeFlyer

hhenryy, 2022-03-16:
I have the same issue.  No sound on Ubuntu 20.04. This is the case for multiple people at my office

rluckom-guardian, 2023-10-27:
Posting in case other people come across this. I experienced what I think is the same issue. A specific laptop running ubuntu 20.04 and QGC 4.2.4 was unable to play sounds from QGC. All other applications (browser, `spd-say`, etc.) were able to play sounds.

The issue was resolved by reinstalling Ubuntu 20.04 and making sure that 3rd-party libraries were enabled during the installation process. In my case this required enabling SecureBoot and setting a secure boot password. I believe that the QT text to speech module relies on something that is missing when ubuntu is installed without the proprietary 3rd-party packages.