mavlink/qgroundcontrol #9584: Issue with Joystick Setup page on QGroundControl

CBWTigerSonKaku, 2021-04-08:
I have QGroundControl ( v4.0.5 ) installed on my Mac ( version 11.2.3), Companion ( v0.0.26 ) installed on the Companion Computer (Raspberry Pi), and Ardusub ( v4.0.3 ) installed on my Pixhawk. I’m sure because I checked http://192.168.2.2:2770/system.

However, when I connect my Logitech F310 Gamepad to my laptop, with the BlueROV2 powered up, the Joystick Setup page on QGroundControl doesn’t become active. I checked my USB ports and all of them are working. I doubt the hardware inside my Logitech F310 Gamepad is the issue. The Joystick Setup page appeared when I plugged in a Sony PlayStation 4 controller. Any thoughts on what the problem could be?

DonLakeFlyer, 2021-04-10:
First thing to do is to instal something like Joystick Show app so you can determine whether OSX is recognizing your joystick at all.

rajeshroy402, 2021-04-16:
Hi, I'm facing something similar to it.
My QGC is not working with Logitech F310 gamepad, I have switched  the vehicle parameter to ` COM_RC_IN_MODE=1` and I'm able to get the 'Joystick' option to calibrate. Even after calibrating it, it's not communicating with the vehicle and keeps on sending me communication error.

cavanaugh-lab-group, 2024-05-10:
did anyone ever figure this out? I am having the same issue

closed 2026-10-04 by github-actions[bot]