mavlink/qgroundcontrol #9586: QGC at MS store

Jai-GAY, 2021-04-09:
Tell us a bit about the feature:
- What problem does it solve? Non admin installation
- What does it do? cannot install
- What flight stacks must it work with? (All, PX4, ArduPilot), all
- Relevant vehicle types? (All, multirotor, fixed-wing, VTOL, submarine, etc.), Multirotor
- Are there other systems that have this feature? IrfanView64
- What communication/integration standards does the feature rely on (e.g. MAVLink commands etc.), None

Any additional context you can provide will make the feature easier to evaluate (e.g. mockups, detailed specification, etc.)
[MS Store](https://www.microsoft.com/en-us/store/apps/windows)
[IrfanView64](https://www.microsoft.com/en-sg/p/irfanview64/9pjz3btl5pv6?activetab=pivot:overviewtab)

The team should immediately look into getting the installation from the MS store for users as some users are using cooperate issued devices where administrator right is not given.

thopiekar, 2021-04-15:
... or alternatively Chocolatey as other opensource software.

rajeshroy402, 2021-04-16:
@Jai-GAY , are you asking for a MS Store App for QGroundControl?

thopiekar, 2021-05-13:
Little follow up on store distribution since I was curious..
https://docs.microsoft.com/en-us/windows/uwp/publish/

kistlin, 2021-06-19:
After reading a bit about the subject, here the current findings.

# Microsoft Store
[image]
>Source: [Top questions](https://developer.microsoft.com/en-us/store/register/)

[How to publish your MSIX package to the Microsoft Store?](https://www.advancedinstaller.com/msix-publish-microsoft-store.html)

# QGroundControl

## [Qt for WinRT](https://doc.qt.io/qt-5.12/winrt-support.html)
`Qt for WinRT` (in Qt 5.12) is now `Qt for UWP` (in Qt 5.15)

>Qt for WinRT allows you to run Qt applications on devices supporting the Windows Store App APIs.

A recompilation against WinRT libraries would be necessary.
But this does not work, because `Qt Serial Port` is not supported in `Qt for WinRT`.

[Qt for UWP](https://doc.qt.io/qt-5/winrt-support.html)
[Qt Add-Ons](https://doc.qt.io/qt-5/qtmodules.html#qt-add-ons)
Platform support (UWP is missing)
- Qt Serial Port: Windows, Linux, macOS, and QNX.

## MSIX package
>MSIX is a modern app package format that provides a universal packaging experience for all Windows apps, WPF, Windows Forms and Win32 apps.

>Source: [Features that require package identity](https://docs.microsoft.com/en-us/windows/apps/desktop/modernize/modernize-packaged-apps#integrate-with-package-extensions)

Packaging Tools from the Microsoft Store: [MSIX Packaging Tool](https://www.microsoft.com/p/msix-packaging-tool/9n5lw3jbcxkf)

### Documentation
[Building an MSIX package from your code](https://docs.microsoft.com/en-us/windows/msix/desktop/source-code-overview)
- What to know before packaging your desktop app
- Packaging your desktop or UWP app in Visual Studio
- CI/CD Pipelines for MSIX Builds and Deployments
- Packaging from the command line
- Extending your MSIX application

[Integrate your desktop app with Windows 10 and UWP](https://docs.microsoft.com/en-us/windows/apps/desktop/modernize/desktop-to-uwp-extensions)

QGroundControl is already packaged with `windeployqt` and should contain everything it needs to run. The only thing missing at a first glance is a manifest file for the MSIX package.

# Thoughts
Going the MSIX route should work and a local proof-of-concept would not need much effort.

I did not look into:
- What is the appropriate Microsoft Store registration (individual or company?) probably company
- Certificate to sign the package
- Integration into CI/CD (link above looks like Azure DevOps integration, not necessarily others)

closed 2026-10-04 by github-actions[bot]