mavlink/qgroundcontrol #9590: Missing Parameter RC_MAP_RATT_SW

cdadswell37, 2021-04-14:
Hi all,

I've found an issue with what seems to be a deprecated parameter preventing QGroundControl from configuring an autopilot as a standard plane or variant thereof. First discovered this issue when trying to build a new version of px4 with a new custom airframe configuration. The firmware built and uploaded fine but QGC throws an error when selecting the flight modes tab of 'missing parameter: RC_MAP_RATT_SW'. Setting the parameter in the airframe config file had no effect, presumably because no prototype for the parameter exists in the px4 firmware anymore, but QGC still seems to be checking for it. Then found the same behaviour for the standard plane airframe too. The px4 firmware version used was 1.11.3 stable, and the QGC version was 4.1.2 and was trialled on both Windows 10 and OSX 11.2.3 and had the same issue on both. The autopilot board used is a Pixhawk 4. I've had a search through the codebases of 1.11.3 and 4.1.2 and there are no mentions at all of RC_MAP_RATT_SW parameter, though I know it exists in previous versions (found it in PX4 1.9.0), so I'm not sure why it's even possible for this error to be thrown. Steps to replicate:
- Run QGC 4.1.2 (installed from the .exe on github)
- Flash PX4 v1.11.3 to a Pixhawk 4
- Set airframe type to standard plane and apply reboot
- Calibrate req. sensors
- Select flight mode tab and error should proc

Thanks very much,
Chris

cdadswell37, 2021-04-20:
Slight update - the missing parameter error only appears to proc with the latest beta 1.12 version of px4, not the 1.11.3 stable. I thought that cloning the repo was giving me the latest stable build, not the latest build full stop. 1.11.3 does still give a 'missing parameters' error on the flight modes tab, but doesn't list any parameters after the error message, and the vehicle seems to arm regardless.

AndrK7, 2021-04-28:
The same issue have QG v4.1.2 FV: 1.12.0 beta.

closed 2021-07-07 by DonLakeFlyer