mavlink/qgroundcontrol #9597: OSD visual configurator as in Inav configurator

cprpolo, 2021-04-18:
Tell us a bit about the feature:
- What problem does it solve? manual osd configuration via parameters is very tedious
- What does it do? same as OnBoard OSD in Mission Planner or OSD in Inav
- What flight stacks must it work with? (All, PX4, ArduPilot) All
- Relevant vehicle types? (All, multirotor, fixed-wing, VTOL, submarine, etc.) All
- Are there other systems that have this feature? yes, INAV and Mission Planner
- What communication/integration standards does the feature rely on (e.g. MAVLink commands etc.) simple OSD, I believe I2C (may be wrong)

ES-Alexander, 2022-03-27:
I was searching about a similar topic after reading a comment elsewhere.

The discussion at #8332 may be relevant/of interest, although that's more about changing QGC's telemetry display/instrument panel to a PFD (primary flight display) style view, whereas this issue seems to be about QGC adding convenient visual interaction options to configure external [OSD boards](https://ardupilot.org/copter/docs/common-osd-boards-on-screen-display.html).

cprpolo, 2022-03-27:
Yes, @ES-Alexander , thats correct, its a matter of making the configuration of the OSD easier

closed 2026-10-04 by github-actions[bot]