mavlink/qgroundcontrol #9601: QGC needs to be developed in an LTS version of QT

mcsauder, 2021-04-20:
Hello,

See #9460 for reference, but the gist of this issue is that for anyone with a current development environment in Windows or Ubuntu Linux, the development setup requirements for QGC not utilizing any kind of LTS version of QT seems simply unsustainable.  It creates a barrier to new developers that is driving contribution away, rather than toward the next better version of QGC.

I have tried migrating toward QT 5.12.10 and newer, as well as the current LTS release of QT6.  So far I have been incapable of making any stock LTS installation or any variation thereof in an updated Windows 10 environment or Ubuntu 20.04 or 20.10 able to compile a working version of QGC in the past 2 weeks.  I am sure I could do it in Ubuntu 16.04, but that is a half decade old...

I am creating this issue to begin the discussion of migrating the development environment toward an LTS version of QT if not the latest, QT6.

Thank you for your suggestions on how to accomplish this.  I will post back to this issue as I make progress toward that objective.

-Mark

DonLakeFlyer, 2021-04-20:
I'm not sure how you come to this conclusion? You seem to be having a problem that is very strange and nobody else is reporting. It seems unique to you. The other problem with using LTS is I believe after Qt 6 LTS is only available to paying customers.

mcsauder, 2021-04-20:
Thanks @DonLakeFlyer , No, QT6, is certainly installable as open source, please uninstall your current version of QT to reproduce: https://www.qt.io/download-open-source?hsCtaTracking=9f6a2170-a938-42df-a8e2-a9f0b1d6cdce%7C6cb0de4f-9bb5-4778-ab02-bfb62735f3e5

DonLakeFlyer, 2021-04-20:
LTS fixes will not be available to open source. As far as I understand you can't get point releases. You'll have to follow the head of absolute latest version to get fixes. But that said, it has nothing to do with the problem you are seeing.

mcsauder, 2021-04-20:
Thanks again, 5.12.10 is LTS.  Don, I'm begging... I've spent weeks with this,... You are the wizard. Please uninstall and reinstall in Windows, and let me know it went ok. :)

mcsauder, 2021-04-20:
Thanks @DonLakeFlyer for all of your help on Slack, I can't say thanks enough.  Here is a Google Doc link that shows the steps we went through troubleshooting: https://docs.google.com/document/d/1xwPnm0GK8GTOexMa-nmU2k4m8vtLzAxPmCqO_aAFEuA/edit?usp=sharing

The issue isn't resolved in the sense of building a Windows 10 executable or installer, but for the purposes of documentation, I wanted to add this comment.

mcsauder, 2021-04-27:
Success! Built successfully on a relatively easy set of modifications to the default qt 5.15.2:
[image]
[image]
[image]

closed 2021-04-27 by mcsauder

mcsauder, 2021-04-27:
> LTS fixes will not be available to open source. As far as I understand you can't get point releases. You'll have to follow the head of absolute latest version to get fixes. But that said, it has nothing to do with the problem you are seeing.

I've checked on this.  LTS fixes are available in open source.

new-user-name, 2021-08-22:
Hi McSauder, can I have a glance at your Googledoc?.. I've sent a request.

new-user-name, 2021-08-22:
Thanks a lot!

mcsauder, 2021-08-22:
Hopefully it is helpful! I had thought I opened access to anyone previously but that obviously wasn't the case. Feel free to add any notes to go with the screenshots if you learn more along the way. It should be a click by click record of what to do.

zdanek, 2022-07-28:
@mcsauder google doc is no longer available. I'm having some issues with QT and I believe your notes might be helpful. Are they somewhere else to look at?

mcsauder, 2022-07-28:
@zdanek , apologies, these links should work.  As you go through each slide would you have any interest in documenting you experience and perhaps it could go into a PX4 doc?

Each slide represents a single mouse click.  Be sure to understand what is taking place with each slide (mouse click).

Linux:
https://docs.google.com/presentation/d/1-wT4mmQwsTtdK5eF6Lf6QBr6FM64aNoqBl3d20RJs_g/edit?usp=sharing

Windows:
https://docs.google.com/presentation/d/1uawmht4T1NequU8GPHlGULuGEgohrGPRW0MFUXkasVU/edit?usp=sharing

zdanek, 2022-08-08:
@mcsauder thank you!
I will look at it. Currently I have set up QT on my Ubuntu, but I'm having issues with Android build so I thought that your notes can help me. So I'm not starting from clean OS without QT.

EngineerAli06, 2022-11-10:
Thank you very much for your help. I've installed and deleted multiple times over the past 4 days.