mavlink/qgroundcontrol #9603: [Feature Request] Full Support for MAV_AUTOPILOT_GENERIC

btlindow, 2021-04-21:
I've been messing around with making a MAVLink autopilot that I want to use QGC to control. I got to the point where I was trying to send a mission start command from QGC, however I noticed that QGC wouldn't send the command because "Guided Mode was not supported on my vehicle". If I glanced at the QGC source correctly, it only looks as if APM and PX4 have Guided Mode supported.

I can get around this by using APM or PX4 in the Heartbeat from my autopilot, but then QGC complains about initial setup and missing parameters which don't necessarily apply to my autopilot.

Is there a reason why this capability doesn't extend to the MAV_AUTOPILOT_GENERIC types?

Thanks for the help!

DonLakeFlyer, 2021-04-21:
Supporting guided commands is firmware specific. There is no standard mavlink spec which allows these features to be implemented in a generic way. Hence there is no mavlink generic support for the. The concept of "Full Support for MAV_AUTOPILOT_GENERIC" is not possible. Only things which are part of mavlink spec can be done with generic.

closed 2021-04-21 by DonLakeFlyer

flybrianfly, 2022-04-23:
I'm currently hitting this issue as well. 

There is the MAVLink command [MAV_CMD_MISSION_START](http://mavlink.io/en/messages/common.html#MAV_CMD_MISSION_START) to command starting the mission. The [MAV_MODE_FLAG](http://mavlink.io/en/messages/common.html#MAV_MODE_FLAG) is encoded in the heartbeat message, so QGC could check that the vehicle is in guided mode first, then send MAV_CMD_MISSION_START. This would allow sensible messages to be displayed to the user, for example:
1. Don't show the mission start unless the vehicle is reporting that it is in guided mode (could add some ARM / ready to fly type flags, i.e. the "ready to fly" flag suggested [here](https://github.com/mavlink/qgroundcontrol/issues/6839). 
2. Send the MAV_CMD_MISSION_START message, display an error message if the vehicle doesn't acknowledge the command.

flexd, 2022-06-07:
I'm hitting this issue today. I'm implementing Mavlink support for something and currently the only way to make this work is either pretend I'm a PX4 or APM (and then suffer/have to implement all/whatever quirks that has), or make changes to and compile QGroundControl for myself to support my vehicle.

When I started to look at the protocol I really expected there to be way for a vehicle to announce what messages/fields in messages/types it supports as part of the connection process. I'm also surprised there is no concept of announcing what GCS has control over what vehicle, if any.