mavlink/qgroundcontrol #9607: Remove DONE button from VTOL_TAKEOFF and VTOL_LAND commands

sanderux, 2021-04-21:
The "Done adjusting" button has no aded value or functionality for these items.
they do add some confusion for users.

I recommend removing them for these items.

Antiheavy, 2021-04-24:
I agree the word "Done" is confusing/jarring when adding mission items like takeoff, survey, etc.  We made the following tweaks to our lightly modified QGC version to be less confusing to our users:

We changed the buttons to [Next] for the Takeoff (and for survey).  This way building a mission feels more familiar to other "wizard" type workflows:
[image]

We kept "Done" on the Landing pattern, but we duplicated the "Loiter clockwise" checkbox on the first step.  This way it really does feel like you are "done" planning the mission when you press [Done]:
[image]

DonLakeFlyer, 2021-04-24:
> The "Done adjusting" button has no aded value or functionality for these items.
> they do add some confusion for users.

For VTOL for example you still need to adjust the takeoff location to be into the wind don't you? With fixed wing we found folks forgetting to do this which is why you go through this step. Otherwise folks were screwing up a leaving along the default orientation.

> I agree the word "Done" is confusing/jarring when adding mission items like takeoff, survey, etc

I do think "Next" is a bit better.

sanderux, 2021-04-25:
> For VTOL for example you still need to adjust the takeoff location to be into the wind don't you? With fixed wing we found folks forgetting to do this which is why you go through this step. Otherwise folks were screwing up a leaving along the default orientation.

Settings the takeoff direction into the wind is something that needs to happen just as setting a landing item or making sure you dont fly into a mountain. I dont see how the "done" button does anything other than confuse and provide an unneeded extra click. Also, not performing the transition into the wind is not as deadly as you might think, just less efficient.

DonLakeFlyer, 2021-05-04:
Ok, if it's not as critical for VTOL this can be changed. For fixed wing it tends to be very critical hence the prompt to prevent nasty screw ups.

sanderux, 2021-07-27:
@DonLakeFlyer any updates on this?