mavlink/qgroundcontrol #9610: return to moving QGC/home

booo, 2021-04-22:
We would like to use QGC on a moving vessel and would like the plane to return to a moving home as soon as we enter RTL or a similar mode. We can't work with a fixed home position because the vessel is moving while the plane is in the air.

As far as I can see there is already an option to stream the QGC position to the vehicle that can be enable in the Settings of QGC.

We would like this to work with ArduPilot/ArduPlane.

I guess we could solve this with an external script too but for ease of use we would like this to be part of QGC.

DonLakeFlyer, 2021-04-24:
QGC doesn't have support for resetting the home position. The code you are referring to is for Follow Me support.

booo, 2021-08-10:
I changed the subject of the issue to address the fact that this is a feature request.

booo, 2022-02-18:
Resolved with ac97ce4

closed 2022-02-18 by booo