mavlink/qgroundcontrol #9611: Joystick feature not working properly

antonio-sc66, 2021-04-22:
## Expected Behavior
Qgroundcontrol should support joysticks https://docs.qgroundcontrol.com/master/en/SetupView/Joystick.html, allowing to arm/disarm and flight the drone.

## Current Behavior
The app behaves strangely and some parts lock up when there's a drone connected to QGC using either a direct Flight controller to GQC connection via USB or when a USB telemetry receiver is connected to the computer. Once the joystick is connected the attitude indicator is still active, but many parts of QGC get stuck or behave strangely, all the components found out to be faulty use Mavlink to send a commands to the drone (arm, mode change, mission, update parameters, mavlink console...).

For example, if you try to arm the drone you get:
`vehicle did not respond to command: MAV_CMD_COMPONENT_ARM_DISAMR`

 In order to get the connection back the joystick has to be disconnected and the app has to be reopened.

This doesn't happen when the drone is connected to QGC via UDP, this has been verified using Gazebo simulation and using a Ethernet UDP Mavlink receiver. In this two cases the app works as expected and the joystick can be used for flights.

## Steps to Reproduce:
1. Open QGC and connect a drone using an USB Telemetry receiver or connect directly via USB to the flight controller
2. Connect the joystick
3. Check that commands can no longer be sent (timeout is reached and QGC returns an error message)

## System Information
- Operating System: Windows 10 and Manjaro
- QGC Version: 4.1.2
- QGC build: stable and built from source
- Flight Controller: Pixhawk 2 and Pixhawk 4
- Autopilot (with version): PX4 1.10.2
- Joystick used: PS4 controller

## Complementary Information
This behaviour has been tested  by [Verónica Panadeiro](https://github.com/vpanadeiro) and me as part of the [ITG](https://itg.es/) team.

Other users have reported what seems to be the same problem using different joysticks in Slack and https://discuss.px4.io/ in the last few days. Examples:
- https://px4.slack.com/archives/CLQFY7D4L/p1618329940129900
- https://discuss.px4.io/t/px4-and-joystick-logitech-3d-pro/22050

## Log Files and Screenshots
- Example [image](https://discuss.px4.io/uploads/default/original/2X/3/3aac74dc6cfaf4d5dbf6264ee7fc8ebe13c87468.jpeg)age,  extracted from: https://discuss.px4.io/t/px4-and-joystick-logitech-3d-pro/22050.

antonio-sc66, 2021-04-22:
On further inspection it looks like this is related to and old issue #9022

rajeshroy402, 2021-04-22:
Thank you @antonio-sc66 for raising this concern.
The above 'current behavior' scenario is same as what I have faced while trying to control my Holybro S500 using Logitech F310 and F710 gamepad.
I have tested this issue using Ubuntu 20.04 LTS, Ubuntu 18.04 and Windows 10 with QGC 4.1.2 and latest daily build version.

hknaio, 2021-04-22:
> > Hi Everyone
> > I have encountered too that issue with same configuration (QGC V4.1.1 Stable). I realized that it is related to new approach writeBytesThreadSafe function in LinkInterface. When Joystick connect to QGC, a new QThread starts. Then it begans to send manual control messages using SerialLink. After Joystick Thread use the QSerialPort::write function, sending mavlink messages stopped and The Link behaves like High Latency Link. I think it is about to thread safety or thread affinity somehow. Because of this, undesired behaivour is emerging. On SITL with UDP Link , there is no issue.
> > There was no problem in old versions of QGC (for example V3.5). I changed the function with old version writeBytes function then it worked correctly.
> > I hope it helps.
> 
> Would you be so kind and give instructions on how to change linkinterface?
> Thank you!

**1-** You have to add this line to LinkInterface.h file signal section

void _invokeWriteBytes(QByteArray);

**2-** Connect this signal to _writeBytes virtual function in LinkInterface constructor

QObject::connect(this, &LinkInterface::_invokeWriteBytes, this, &LinkInterface::_writeBytes);

**3-** Change writeBytesThreadSafe function with this:

void LinkInterface::writeBytesThreadSafe(const char *bytes, int length)
{
     emit _invokeWriteBytes(QByteArray(bytes, length));
}

It should work.

Note: I do not claim that this is the best solution. I'm working on it. But I'm sure it's about SerialLink not UDPLink. It occurs when Joystick and SerialLink Threads work together.

jturner-anduril, 2021-04-29:
I can confirm seeing this on Windows 10 and Android (both version 4.1.2)

kadirozgunn, 2021-05-09:
I can see this issue with QGC 4.1.2 and 4.1.3  on Windows 10 and MacOS BigSur. I use xBox 360 controller. I can calibrate controller however when I try to give command it gives error.

DonLakeFlyer, 2021-05-09:
@jaxxzer You guys are most familiar with the joystick code I think. Do you have anyone that could take a look?

hknaio, 2021-05-12:
i have found a temporary solution on here #9022 

https://github.com/mavlink/qgroundcontrol/issues/9022#issuecomment-824929620

may be it helps.

kadirozgunn, 2021-05-13:
@AlexKlimaj i had that issue with QGC 4.1.2 and 4.1.3 stable builds with xbox 360 controller. I would be happy If you test that builds.