mavlink/qgroundcontrol #9613: Android UI show nothing after building

shmm91, 2021-04-23:
# This system is for posting bugs or feature requests ONLY. Support questions will be automatically closed with no response.
# For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

----
# To post a bug report
When posting bug reports, include the following informaiton:
- Ground station operating system.
- QGroundControl version and build (daily, stable, self-built from source, etc.)
- **If you are using a Stable build which is not the latest, first step is to install latest Stable build and try again prior to entering a bug report.**
- Autopilot board: Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.
- Autopilot firmware (e.g. PX4, ArduPilot, custom) and version.
- Exact steps to reproduce the problem. Starting from booting QGroundControl to when the problem occurs.

**All this template text should be deleted before creating your issue.**

I build QGroundControl in android mode.UI show like this,what's wrong with my building?
[image]

louistHD, 2022-04-29:
I'm having exactly this issue. Did you fix it? Any ideas @DonLakeFlyer ?

booo, 2022-09-15:
Several people check the current android builds and did not report any problems. I assume this problem is resolved. Please reopen with additional information if the problem persists.

closed 2022-09-15 by booo