mavlink/qgroundcontrol #9626: Address Sanitizer: `new-delete-type-mismatch`

keith-bennett-airmap, 2021-04-30:
Hi,

As [I mentioned in a previous pull request](https://github.com/mavlink/qgroundcontrol/pull/9595#issuecomment-824974766), I want to add some debugging tools that automatically detect problems; one such tool is Address Sanitizer. I don't want to commit / offer a pull request unless and until the tools report that the code is clean without needing any massaging: I don't think it'd be fair to pollute the CI system with issues that are outstanding.

To that effect, Address Sanitizer has demonstrated a bug that I can't quickly resolve. The bug is in QML code but my experience is deeply in C++ instead. From the stack trace, I don't see any stack frames within QGC code so it's possible the bug is in upstream Qt. It could be that the allocation and deletion of the object happens only during Qt code but that QGC QML has incorrect handling of the complained object. I'm not sure.

## Expected Behavior
I expect QGroundControl to never cause Address Sanitizer to trigger an abort.

## Current Behavior
QGroundControl, when built with Address Sanitizer, may abort after detecting a `new-delete-type-mismatch` error.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Build QGC with Address Sanitizer. It's not a default build type, so please follow instructions (at end of this ticket) to add it to your build type.
2. Run the Address Sanitizer-instrumented QGC. Be sure to use the "Start Debugging" feature of Qt Creator so that Address Sanitizer output will be caught and displayed.
3. Add a MockLink interface. Connect the interface. Wait for the vehicle to appear and then start planning. Just add waypoints until the crash happens. At the end of this ticket there is an animated GIF demonstrating the problem.
4. When the crash happens, you'll see the summary describing the exact problem that Address Sanitizer detected and a stack trace of relevant events. It may offer a workaround. In this case, we can add **`new_delete_type_mismatch=0`** to the **`ASAN_OPTIONS`** runtime environment variable mentioned during setup.

## System Information
When posting bug reports, include the following information
- Operating System: Pop! OS 20.04, Cinnamon Desktop
- QGC Version: master c656d372e01fa69b5e570f816dcf39487e140659
- QGC build: self-built from source
- Qt: 5.12.10, GPLv3 built from source
- Flight Controller:  none (MockLink)
- Autopilot (with version): none (MockLink)

## Detailed Description
I suspect that either a destructor isn't marked as virtual (and so the deleted type is smaller than the allocated type) or that QML has a type confusion error. I don't have sufficient knowledge of QML to debug this in a timely fashion. It's possible that the bug lies in upstream Qt library but I suspect that the bug is in QGC's handling of animations with geographical coordinates.

A non-virtual destructor _should_ cause a compiler warning which I haven't observed. So if that's the cause then the missing virtual destructor is in Qt library code.

The bug might be benign with no adverse affects (eg, if the sliced type only contains plain data), it might cause memory leaks (eg, if the sliced type owns allocations), or it might cause deadlocks and/or crashes (if the sliced type or its users expect to be able to interact with the destroyed type).

In my experience, new/delete type mismatches cause subtle and difficult-to-diagnose bugs.

I did do a little bit of digging into the problem.

1. Stack trace points to [the allocation of a `QGeoCoordinatePrivate` object](https://github.com/qt/qtlocation/blob/v5.12.10/src/positioningquick/qquickgeocoordinateanimation.cpp#L234) which [is derived from `QSharedData`](https://github.com/qt/qtlocation/blob/v5.12.10/src/positioning/qgeocoordinate_p.h#L59-L76).
2. [`QSharedData` does not declare a destructor](https://github.com/qt/qtbase/blob/v5.12.10/src/corelib/tools/qshareddata.h#L55-L66), so it's certainly not `virtual`. However, `QSharedData` is used _everywhere_ in Qt internals so if its missing virtual dtor were _the_ problem then Qt would be a _lot_ more noisy to Address Sanitizer!
3. The destructor stack points to [the class declaration of `QQuickStateAction`](https://github.com/qt/qtdeclarative/blob/v5.12.10/src/quick/util/qquickstate_p.h#L67) even though it's in the destructor for it. So that means it's in an auto-generated destructor. I suspect it's destroying one of the member `QVariant` types.

So from the destructor I  can only start to make educated guesses. It would take much longer amounts of time to diagnose than I can assign to this problem. One unanswered question still remains: is it a bug in Qt or is QGC incorrectly using QML and triggering the problem? I don't have the answer.

## Log Files and Screenshots

- Instructions for adding Address Sanitizer build target:
  - Create a new build type from the project build screen
  [image]
  - I named it "ASan". You could name it "Address Sanitizer" if you want, but your Qt tooling will likely truncate the name when being displayed elsewhere.
  [image]
  - Add **`CONFIG += sanitizer CONFIG += sanitize_address`** to the build options. I tried using GCC's `fsanitize=address` here and that _does not work_. If you're feeling frisky you can try [other sanitizers](https://stackoverflow.com/a/39041217/1111557). I used Thread Sanitizer to solve a CI stability issue, but TSan is _extremely noisy_ in QGC. Many problems reported by TSan are real but outside of the scope of this ticket.
  [image]
  - QGC, like most Qt projects, is very leaky. If you don't want Address Sanitizer to complain about the leaks then you can disable the check for that by adding **`ASAN_OPTIONS`** environment variable and setting it to **`detect_leaks=0`**. A lot of the leaks can be fixed but they're outside of the scope of this issue.
  [image]
  - The new build type should be selected automatically. But if not, you should ensure that it is selected.
  [image]
  - Finally, you can build. If it doesn't build successfully then you have probably done something wrong.
  [image]

- Address Sanitizer error report:
  ```
  =================================================================
  ==92077==ERROR: AddressSanitizer: new-delete-type-mismatch on 0x604000957fd0 in thread T0:
    object passed to delete has wrong type:
    size of the allocated type:   48 bytes;
    size of the deallocated type: 32 bytes.
      #0 0x7ffff7681025 in operator delete(void*, unsigned long) (/lib/x86_64-linux-gnu/libasan.so.5+0x111025)
      #1 0x7ffff1cda65c in customTypeDestructor kernel/qmetatype.cpp:2021
[... 29 lines cut ...]
      #31 0x7fffe9d04f09 in xcbSourceDispatch /home/qt/work/qt/qtbase/src/plugins/platforms/xcb/qxcbeventdispatcher.cpp:105
      #32 0x7ffff608317c in g_main_context_dispatch (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x5217c)
      #33 0x7ffff60833ff  (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x523ff)
      #34 0x7ffff60834a2 in g_main_context_iteration (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x524a2)
      #35 0x7ffff1d1b8fe in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) kernel/qeventdispatcher_glib.cpp:422
      #36 0x7ffff1cc2da9 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) kernel/qeventloop.cpp:225
      #37 0x7ffff1ccb7ef in QCoreApplication::exec() kernel/qcoreapplication.cpp:1389
      #38 0x5555560cefd1 in main /home/kbennett/src/public/qgroundcontrol/src/main.cc:401
      #39 0x7ffff15370b2 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x270b2)
      #40 0x55555587090d in _start (/home/kbennett/src/public/qgroundcontrol/build/build-qgroundcontrol-Desktop_Qt_5_12_10_GCC_64bit-ASan/staging/QGroundControl+0x31c90d)
  
  0x604000957fd0 is located 0 bytes inside of 48-byte region [0x604000957fd0,0x604000958000)
  allocated by thread T0 here:
      #0 0x7ffff767f947 in operator new(unsigned long) (/lib/x86_64-linux-gnu/libasan.so.5+0x10f947)
      #1 0x7ffff0d9dbe9 in QQuickGeoCoordinateAnimation::setTo(QGeoCoordinate const&) /home/qt/work/qt/qtlocation/src/positioningquick/qquickgeocoordinateanimation.cpp:234
      #2 0x7ffff55371d3 in QQuickWindowPrivate::sendFilteredMouseEvent(QEvent*, QQuickItem*, QQuickItem*) items/qquickwindow.cpp:3108
  
  SUMMARY: AddressSanitizer: new-delete-type-mismatch (/lib/x86_64-linux-gnu/libasan.so.5+0x111025) in operator delete(void*, unsigned long)
  ==92077==HINT: if you don't care about these errors you may set ASAN_OPTIONS=new_delete_type_mismatch=0
  ==92077==ABORTING
  11:58:16: Debugging has finished
  ```

- Problem demonstration GIF
  [image]

DonLakeFlyer, 2021-05-04:
Thanks @keith-bennett-airmap. I'm back in town now, I'll try to get to this as soon as I dig out from being gone.

DonLakeFlyer, 2021-07-20:
@keith-bennett-airmap this is all clean now right?

keith-bennett-airmap, 2021-07-20:
@DonLakeFlyer 

No, this still happens as of c0e3a4e96e6d85ecf5f25758b77bd7936a4ccfc1

keith-bennett-airmap, 2021-07-20:
[image]

[2021-07-20-c0e3a4e96e6d85ecf5f25758b77bd7936a4ccfc1.txt](https://github.com/mavlink/qgroundcontrol/files/6850914/2021-07-20-c0e3a4e96e6d85ecf5f25758b77bd7936a4ccfc1.txt)

DonLakeFlyer, 2021-07-20:
From just looking at the stack trace this looks like a Qt bug associated with flicking the map. There isn't any QGC code in the stack.

keith-bennett-airmap, 2021-07-20:
OK I thought as much but wanted a second pair of eyes. I'll go poke some more at Qt

DonLakeFlyer, 2021-07-20:
There have been problems with map flicking before. I've created repros of the bugs with simple Qml apps and standard qt map control, created Qt bugs to report and watched them sit there for years without ever being fixed.

keith-bennett-airmap, 2021-07-20:
Do you have some bug reports on qt.io to reference?

Since you mentioned it looks like a problem with flicking in maps, I poked around the Qt examples with map flicking. Indeed it's trivially reproducible there. I created a Qt bug report -> https://bugreports.qt.io/browse/QTBUG-95294

DonLakeFlyer, 2021-07-20:
Search for `reporter = DonLakeFlyer` and you should find a few