mavlink/qgroundcontrol #9627: Crash QGroundControl when open Telemetry log file

WTPENGUIN, 2021-05-03:
## Expected Behavior
QGC show Telemetry Log normaly

## Current Behavior
QGC crashes with a high probability.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Show Telemetry Log Replay Status Bar
2. Open *.tlog file created by QGC
3. QGC crashes with a high probability.

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10
- QGC Version: 4.1.2
- QGC build: stable
- Flight Controller: Just Open *.tlog file
- Autopilot (with version): Just Open *.tlog file

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)

[QGroundControl.exe.4560.zip](https://github.com/mavlink/qgroundcontrol/files/6412873/QGroundControl.exe.4560.zip)
[QGroundControl.exe.8456.zip](https://github.com/mavlink/qgroundcontrol/files/6412901/QGroundControl.exe.8456.zip)
[QGroundControl.exe.11628.zip](https://github.com/mavlink/qgroundcontrol/files/6412906/QGroundControl.exe.11628.zip)

- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

kistlin, 2021-06-12:
When replaying logs I have the same issue, but all the time. I'm using a cmake build of the current master.

When debugging I get the following callstack
```
mavlink_frame_char_buffer(__mavlink_message *,__mavlink_status *,unsigned char,__mavlink_message *,__mavlink_status *) mavlink_helpers.h:595
mavlink_frame_char(unsigned char,unsigned char,__mavlink_message *,__mavlink_status *) mavlink_helpers.h:893
mavlink_parse_char(unsigned char,unsigned char,__mavlink_message *,__mavlink_status *) mavlink_helpers.h:971
LogReplayLink::_findLastTimestamp() LogReplayLink.cc:243
LogReplayLink::_loadLogFile() LogReplayLink.cc:273
LogReplayLink::run() LogReplayLink.cc:120
QTextStream::realNumberPrecision(void)const  0x00007ffefcb6aa8b
BaseThreadInitThunk 0x00007fff79097034
RtlUserThreadStart 0x00007fff79662651
```

In `mavlink_frame_char_buffer` line 595 `rxmsg->len = 0;` we access an address that is not a valid location.

Now where do we get `rxmsg` from.
This is returned from `mavlink_get_channel_buffer(chan)` in `mavlink_helpers.h` line 893.

```
MAVLINK_HELPER mavlink_message_t* mavlink_get_channel_buffer(uint8_t chan)
{
	
#ifdef MAVLINK_EXTERNAL_RX_BUFFER
	...
#else
	static mavlink_message_t m_mavlink_buffer[MAVLINK_COMM_NUM_BUFFERS];
#endif
	return &m_mavlink_buffer[chan];
}
```

In my case the else part is active and we have an array of length `MAVLINK_COMM_NUM_BUFFERS` = 16.
But my `chan` passed in is 205. Which leads to an invalid pointer at `&m_mavlink_buffer[chan];`

Now `chan` comes from `LogReplayLink::_findLastTimestamp` which passes `_mavlinkChannel` as `chan` argument to `mavlink_parse_char`.

As it seems `_mavlinkChannel` is just not initialized and never set later during execution. And therefore sometimes crashes and sometimes works.

After initializing it with `0` I can replay the logs.

kistlin, 2021-06-12:
I opened the PR #9707 which contains the changes that fixed it for me. Maybe someone can reproduce it and test the fix.

WTPENGUIN, 2021-06-12:
After applying PR, I've tested it with the Telemetry logs I have and it doesn't seem to be a problem. Thank you, @kistlin

PR #9707 (2021-06-13): initialize member variables of class LogReplayLink
fixes #9627
commit 06160182 (2021-06-13): initialize member variables of class LogReplayLink

- fixes #9627
closed 2021-06-13 by DonLakeFlyer