mavlink/qgroundcontrol #9629: libairmap-qt.so.1.1.0 not found

julianoes, 2021-05-04:
## Expected Behavior
I would like to just be able to build QGC and run it, so I can do development.
I use these steps:
```
mkdir build && cd build
~/Qt/5.12.9/gcc_64/bin/qmake -r ../qgroundcontrol.pro CONFIG+=force_debug_in
make -j8
```

## Current Behavior
It builds but doesn't start:

```
./staging/QGroundControl: error while loading shared libraries: libairmap-qt.so.1.1.0: cannot open shared object file: No such file or directory
``` 

Even though the file is there:
```
./staging/Qt/libs/libairmap-qt.so.1.1.0
./libs/airmap-platform-sdk/linux/Qt.5.12/libairmap-qt.so.1.1.0
```

## Steps to Reproduce:
See above.

## System Information
When posting bug reports, include the following information
- Operating System: Arch Linux
- QGC Version: master (2b2d9d826cfd4353c64b98312836b763cadbfa66)

booo, 2021-05-07:
Did you try something like this: `LD_LIBRARY_PATH=./staging/Qt/libs ./staging/QGroundControl`?

julianoes, 2021-05-07:
@booo thanks. Yes I use that until it's fixed.

jennerl, 2021-05-07:
@julianoes `LD_LIBRARY_PATH` is set to include libs in `./staging/Qt/libs` via https://github.com/mavlink/qgroundcontrol/blob/master/deploy/qgroundcontrol-start.sh#L3

Are you using that script to start QGC?

julianoes, 2021-05-08:
Do I need to use that script? What's wrong with just starting the executable directly? Sorry, I'm playing a bit ignorant here on purpose trying to understand why we need such things.

booo, 2021-05-11:
For an explanation of the LD_LIBRARY_PATH please follow this link: https://stackoverflow.com/a/7148086

We build libraries and link against them. If the shared libraries are not in your env they can't be found when you start the script. Therefore use the LD_LIBRARY_PATH to add them to your env or use the start.sh script.

I think you can close this ticket. It's not a bug.

closed 2021-05-11 by julianoes