mavlink/qgroundcontrol #9636: Cannot calibrate radio, even though valid(?) mavlink messages are being received

teyrana, 2021-05-09:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues--

## Expected Behavior
Please tell us what should happen as detailed as possible.

## Current Behavior
Please describe what happens instead of the expected behavior.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1.
2.
3.
4.

## System Information
When posting bug reports, include the following information
- Operating System:  Ubuntu 20.04.2 / Kernel 5.4.0-72
- QGC Version: [ AppImage Version ]( https://s3-us-west-2.amazonaws.com/qgroundcontrol/latest/QGroundControl.AppImage )
- QGC build: Development HEAD:78cf9bbe6 2021-05-06 10:49:59 -0700
- Flight Controller: Pixhawk 4 + Spektrum Rx 4651t + Spektrum Tx DX6e
- Autopilot (with version): PX4 / Development branch: https://github.com/PX4/PX4-Autopilot/pull/17555

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots

The following screenshots show that RC input is making it to GQC, with roughly the values I would expect.
(note: the values *I* expect. I don't know what values QGC expects.)

[image]
[image]

DonLakeFlyer, 2021-05-09:
> The following screenshots show that RC input is making it to GQC

What two states do those two screenshots represent? And what channels are what on your radio?

DonLakeFlyer, 2021-05-09:
Just guessing but if the first image is initial state with sticks centered and channel 1 is roll something seems wrong if the centered roll value is 1705 instead of closer to 1000 like the other sticks.

teyrana, 2021-05-09:
@DonLakeFlyer 

Sorry, those two screenshots correspond to right stick down-left and down-right.    (As close as I can figure to the last two steps in the calibration procedure... would it help if I had photos of the calibration steps?  or video? 

channel 2 is right stick up/down. 
channel 1 is right stick left/right.

Center on channels 1 and 2 is both 1024.   Min value is ~= 343 and max is a bit over 1700.  For both channels.

DonLakeFlyer, 2021-05-11:
Turn on console logging forRadioComponentControllerLog and attach the log here. https://docs.qgroundcontrol.com/master/en/SettingsView/console_logging.html

teyrana, 2021-05-13:
## Radio Cal Sliders at step g: (without input)
[image]

## Radio Cal Sliders at Step h: (with transmitter input)
[image]

## Console Log:
[qgroundcontrol.log.txt](https://github.com/mavlink/qgroundcontrol/files/6469855/qgroundcontrol.log.txt)

teyrana, 2021-05-13:
Note: if it ends up that the radio input is just out of range that would still be useful.   If I know the appropriate ranges, it can be fixed upstream.

DonLakeFlyer, 2021-05-13:
I assume you really are setup to want Mode 1?

DonLakeFlyer, 2021-05-13:
I you don't know what Mode 1 is then you don't want it and should be using Mode 2.

teyrana, 2021-05-13:
No, I just went with the default mode.  Neither mode worked, though, and both freeze at the same step.

Wouldn't both of them need to work, thought?   I mean, **_I_** don't need mode 1, but I assume someone else with the Spektrum receiver would?

DonLakeFlyer, 2021-05-13:
Mode 1 isn't going to work on a non Mode 1 transmitter since the throttle position will be wrong. You need to pick Mode 2 and then redo the steps and attach the log.

DonLakeFlyer, 2021-05-13:
The default should be mode 2 not 1.

teyrana, 2021-05-15:
a) The QGC docs don't mention mode 1 /2 ... at all:
https://docs.qgroundcontrol.com/master/en/SetupView/Radio.html

b) my transmitter has 4 modes, so calibrating in mode 1 or 2 seems like a legit request? 
    I'm not implementing this just for me... I'm implementing this for anyone who wants to buy those spektrum receivers :)

teyrana, 2021-05-17:
If you wish to debug in mode 2 here you go:  (both transmitter and QGC are in mode 2):
[qgroundcontrols-mode2.log.txt](https://github.com/mavlink/qgroundcontrol/files/6490949/qgroundcontrols-mode2.log.txt)

If you want to debug in mode 1, here's that log, where the transmitter and QGC are in mode 1:
[qgroundcontrols-mode1.log.txt](https://github.com/mavlink/qgroundcontrol/files/6490991/qgroundcontrols-mode1.log.txt)

> Mode 1 isn't going to work on a non Mode 1 transmitter ...

The transmitter [supports mode 1, mode 2, mode 3, and mode 4](https://www.spektrumrc.com/Products/Default.aspx?ProdId=SPMR6650).

> You need to pick Mode 2 . . .

I still don't follow... Surely you're not saying the Spektrum rx / tx pair will ONLY calibrate in one of the modes !?
If so: (a) let's add that to the documentation!  (b) if you know what the issue is, I'll see if I can fix it on the px4 side.

----

I would much rather talk about how the logs show calibration on the throttle axis just fine.  The problems only start on the roll step --  As you can see in the logs, there's no obvious error, it just fails to detected the axis... 

Specifically, I see a log line like:  `_setHelpImage "calibration/mode2/radioRollRight.png"`
following later by a line that says: `"_inputStickDetect starting settle wait, function:channel:value 0 0 584"` 
followed by: `"_inputStickDetect settle complete, function:channel:value:reversed 0 0 343 true"`

But after: `"_setHelpImage "calibration/mode2/radioRollLeft.png""`
we don't see any further `_inputStickDetect` log lines.    => this corresponds to the undetected left-roll input.
This is in spite of detected roll axis values that should trigger it: ` "Raw value 0 1705"`

DonLakeFlyer, 2021-05-17:
If you look at your mode 2 logs you can see the _inputStickDetect for channels 2, 3, 0 which leads me to believe your roll channel (which is the last to cal) should come through on channel 1. But if you look at the raw values for channel 1 at the end part of the log, channel 1 never really moves beyond a center value. QGC needs a delta of at least 300 PWM change to detect a stick. Hence it stops at detecting your roll stick since no sticks are moving.

I'm am not saying anything about modes not being supported. I am only pointing to debugging in mode 2 which is simpler.

DonLakeFlyer, 2021-05-17:
I also discovered that there is some missing logging with respect to the step that happens after _inputStickDetect which is a minimum determination. I'll add that logging which will be helpful. But isn't really needed in this specific case since it is _inputStickDetect which is failing.

teyrana, 2021-05-18:
> _inputStickDetect for channels 2, 3, 0 which leads me to believe your roll channel (which is the last to cal) should come through on channel 1.

Oh, that's very interesting.  I could certainly swap channels 0 & 1.

Do you have any insight why it's detecting the right-roll, but not the left roll, though?

DonLakeFlyer, 2021-05-19:
It's not detecting any roll at all. The values for roll channel 1 don't change.

DonLakeFlyer, 2021-05-19:
I would check the raw mavlink coming in for channel 1 which is chan2_raw in the mavlink inspector and move it full right/left to see if you get good values at each extreme. If you don't then something is wrong with your radio setup not QGC.

teyrana, 2021-05-22:
We do not yet have a clear communication of the issue -- for more detail, here are screenshots of the entire window: 

## Before step g (calibrate roll right): 

[image]

Please note how before step g, The roll axis is "Not Mapped".
- This is also noted in the mode-2 log, attached below, at lines 2104-2106

## ... and before step h (calibrate roll left)

[image]

Please note how after step g, The roll axis is now mapped.  (it detects the right extreme) 
But during step h, it cannot detect the left extreme.   These two data points suggest that the roll channel is partially calibrated.
This is also noted in the mode-2 log, attached below, after line 2245
  
## Updated log

Mode 2, raw transmitter channel numbers

[qgroundcontrols-mode2-20210522.log.txt](https://github.com/mavlink/qgroundcontrol/files/6526781/qgroundcontrols-mode2-20210522.log.txt)

  - The right extreme of "Raw value 1 343" / `chan2_raw` can be seen in the log at line 2106
  - The left extreme of "Raw value 1 1705" / `chan2_raw` can be seen in the log at line 2245
 
## Channel mapping:
  - all channel re-mapping is removed.  This has no visible effect on calibration.

## Verify flow of channel values: 

As requested:
> check the raw mavlink coming in for channel 1 which is chan2_raw in the mavlink inspector and move it full right/left to see if you get good values at each extreme.

Verified.   The mavlink data: (`RC_CHANNELS`::`chan2_raw` )  ranges from 343 (right) - 1705 (left)

This is visible in the MAVLink Inspector tab: 
[image]
This specifically show the input from rx -> tx -> px4 -> mavlink -> qgc  during step h.  (the problematic step)

DonLakeFlyer, 2021-05-22:
Ok, my eyes must have been crossing over looking at all the numbers. I see it now. I have to look at the code some more but my initial guess is that there is QGC bug which relates to the roll channel being reversed. I'd guess if you make the low PWM be stick left and high PWM be stick right it will work correctly. Or at least that's a definite maybe!

DonLakeFlyer, 2021-05-22:
I tried simulating the reversed roll channel and of course it all seems to work fine for me. I've added better/missing logging to radio cal in the referenced pull. That should help figure this out.

DonLakeFlyer, 2021-05-22:
Oh, wait. I see it now. PWM values are expected to range from 1000<->2000. With 1500 being centered. But your radio is pushing out values which seems to be in the 0<->2000 range which seems odd. I've never seen this before. Is this some sort of new radio thing?

teyrana, 2021-05-22:
>   PWM values are expected to range from 1000<->2000. With 1500 being centered. But your radio is pushing out values
> which seems to be in the 0<->2000 range, with 1000ish being centered which seems odd. 

Ahhh, it's a bit more complicated than that -- px4 already did some conversion, but ... not enough for QGC, apparently.
However, once I scaled & recentered to [1000,2000] it seems to calibrate okay.   We'll see how controllability is, now :)

https://github.com/teyrana/PX4-Autopilot/commit/1b450a9fde84e4f237b211f9f7579a4f2a516d7b

>  I've never seen this before. Is this some sort of new radio thing?

"Newish".  It's the Spektrum receiver SRXL format.   APM is also working supporting it, so Im' a little surprised this hasn't come up before, but .... /shrug

https://github.com/SpektrumRC/SRXL2/blob/master/Docs/SRXL2%20Specification.pdf

The short version is that the interval [300-1700] is not a native encoding, either.   The SRXL2 raw encoding is: [ 0x2aa0, 0xd554] ~= [10912, 54612]   ... and the rest is just conversion.

teyrana, 2021-05-22:
## Resolution:

It's still mysterious why the throttle and yaw channels could calibrate on the [343,1700] interval, but the roll channel could only halfway calibrate....  

But the good news is, this is no longer a blocker for the PX4 pr.

@DonLakeFlyer thanks for all your help, I couldn't have fixed it without your help, and hard work :)

DonLakeFlyer, 2021-05-23:
> It's still mysterious why the throttle and yaw channels could calibrate on the [343,1700] interval, but the roll channel could only halfway calibrate....

Just a fluke of exercising code in unexpected ways. The PWM inputs are not validated to an expected valid range.

closed 2021-05-23 by DonLakeFlyer

teyrana, 2021-05-23:
> The PWM inputs are not validated to an expected valid range.

For future reference, this particular hardware is digital -- i.e. a serial-port-digital-protocol.  In this case, it's transmitted on-vehicle as 16-bit values (no PWM involved)

Anyway, is there an appropriate place we can document this expectation (that QGC expects values from 1000-2000?)

DonLakeFlyer, 2021-05-25:
There are docs for Radio Cal which can be modified. That said, this sort of thing seems pretty uncommon since I would expect an RC radio to output values/PWM in a standard servo range which is 1000-2000.

teyrana, 2021-05-27:
Well, as you wish.