mavlink/qgroundcontrol #9637: Set RC to Param - mismatch values

diablo1281, 2021-05-10:
I am testing "Set RC to Param" function in my custom autopilot firmware. It seems like values "Scale" and "Center value" are switched.

In version 3.5.6 it looks ok, but since version 4.x.x it doesn't work properly.

DonLakeFlyer, 2021-05-20:
I looked at the code and as far as I can tell they are correct. Do you have some steps to reproduce which show the problem?

diablo1281, 2021-05-27:
Here are some screenshots with the way we found it:
1. During debug of FCU firmware set breakpoints for message PARAM_MAP_RC.
2. Open QGroundControl 3.5.6 and set any param to RC. (screenshot below)
[image]
3. In debugger we hit a breakpoint and can check values for reviced message. (screenshot below) 
[image]
4. Next repeat steps above in QGC 4.1.0+. In screenshots below you can see values `scale` and `value0` are switched.
[image]
[image]

PR #9684 (2021-06-11): RC to Param: Fix swapped values
Fix #9637
closed 2021-06-11 by DonLakeFlyer