mavlink/qgroundcontrol #9667: QGC crashes when "Reboot Vehicle" button is pressed after calibration steps.

kressinluiz, 2021-05-21:
## Expected Behavior
To finish calibration, the "Reboot Vehicle" button should reboot the vehicle, close the window and regain comms when the vehicle is up again.

## Current Behavior
When the "Reboot Vehicle" button is pressed after the calibration steps, QGC crashes. 

[image]
Also the reboot button goes of the screen after compass calibration.
[image]

## Steps to Reproduce:
1. Go to Application Settings -> Comm Links
2. Connect to Skydroid RC T10 via Bluetooth.
3. Go to Vehicle Setup -> Sensors
4. Go to Accelerometer or Compass and proceed with the calibration steps as usual.
5. When finished, click on Reboot Vehicle.
6. QGC crash.

## System Information
When posting bug reports, include the following information
- Operating System: Android 9
- QGC Version: 4.1.1 32 bit
- QGC build: Stable and Daily (246d5742b 2021-01-17) from Google Play Store
- Flight Controller: Pixracer
- Autopilot (with version): Ardupilot copter 4.0.3

## Detailed Description
I first noticed this on a custom build of v4.1.3 and got the following log:
[image]

Thanks in advance!

DonLakeFlyer, 2021-05-28:
Can you try a newer daily. Latest versions aren't avialable on the store. There are links to APKs you can download in the docs: https://docs.qgroundcontrol.com/master/en/releases/daily_builds.html

kressinluiz, 2021-05-28:
Hello @DonLakeFlyer. I've tried with a newer daily (3b0d3b234 2021-05-27 32 bit) and got the same results when "Reboot Vehicle" button is activated: the vehicle reboots but QGC crashes.

When I was testing I noticed another bug. Should I create another issue? It goes like this:

1. While connected on vehicle, go to Vehicle Setup -> Sensors and click on any of the calibration options.
2. While the right side panel is open, click on the "Back" button
[image]
3. QGC goes to Fly View but the right side panel do not close and it is not possible to close it by clicking on any button. Only available option is to restart QGC.
[image]

Thanks for your help. 
Have a nice day!

closed 2026-10-04 by github-actions[bot]