mavlink/qgroundcontrol #9669: build error: argument 2 of `__atomic_load` must not be a pointer to a `volatile` type

acxz, 2021-05-22:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Please tell us what should happen as detailed as possible.
The compilation process should succeed.

## Current Behavior
Please describe what happens instead of the expected behavior.
QGroundControl fails to build.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. git clone ...
2. mkdir build
3. qmake ..
4. make

## System Information
When posting bug reports, include the following information
- Operating System: Archlinux
- QGC Version: 4.1.3
- QGC build: building from source
- Flight Controller: N/A
- Autopilot (with version): N/A

## Detailed Description
Provide further details about your issue/bug.

<details>
  <summary>Log</summary>

  ```
In file included from /usr/include/glib-2.0/glib/gthread.h:32,
                 from /usr/include/glib-2.0/glib/gasyncqueue.h:32,
                 from /usr/include/glib-2.0/glib.h:32,
                 from /usr/include/gstreamer-1.0/gst/gst.h:27,
                 from /home/acxz/vcs/git/github/acxz/pkgbuilds/qgroundcontrol/src/qgroundcontrol-4.1.3/libs/qmlglsink/gst-plugins-good/ext/qt/gstqtglutility.h:24,
                 from /home/acxz/vcs/git/github/acxz/pkgbuilds/qgroundcontrol/src/qgroundcontrol-4.1.3/libs/qmlglsink/gst-plugins-good/ext/qt/gstqtglutility.cc:25:
/home/acxz/vcs/git/github/acxz/pkgbuilds/qgroundcontrol/src/qgroundcontrol-4.1.3/libs/qmlglsink/gst-plugins-good/ext/qt/gstqtglutility.cc: In function ‘GstGLDisplay* gst_qt_get_gl_display()’:
/usr/include/glib-2.0/glib/gatomic.h:117:19: error: argument 2 of ‘__atomic_load’ must not be a pointer to a ‘volatile’ type
  117 |     __atomic_load (gapg_temp_atomic, &gapg_temp_newval, __ATOMIC_SEQ_CST); \
      |     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/glib-2.0/glib/gthread.h:260:7: note: in expansion of macro ‘g_atomic_pointer_get’
  260 |     (!g_atomic_pointer_get (location) &&                             \
      |       ^~~~~~~~~~~~~~~~~~~~
/home/acxz/vcs/git/github/acxz/pkgbuilds/qgroundcontrol/src/qgroundcontrol-4.1.3/libs/qmlglsink/gst-plugins-good/ext/qt/gstqtglutility.cc:65:7: note: in expansion of macro ‘g_once_init_enter’
   65 |   if (g_once_init_enter (&_debug)) {
      |       ^~~~~~~~~~~~~~~~~
make: *** [Makefile:134434: gstqtglutility.o] Error 1
make: *** Waiting for unfinished jobs....
In file included from /usr/include/glib-2.0/glib/gthread.h:32,
                 from /usr/include/glib-2.0/glib/gasyncqueue.h:32,
                 from /usr/include/glib-2.0/glib.h:32,
                 from /usr/include/gstreamer-1.0/gst/gst.h:27,
                 from /usr/include/gstreamer-1.0/gst/video/video.h:23,
                 from /home/acxz/vcs/git/github/acxz/pkgbuilds/qgroundcontrol/src/qgroundcontrol-4.1.3/libs/qmlglsink/gst-plugins-good/ext/qt/gstqsgtexture.cc:28:
/home/acxz/vcs/git/github/acxz/pkgbuilds/qgroundcontrol/src/qgroundcontrol-4.1.3/libs/qmlglsink/gst-plugins-good/ext/qt/gstqsgtexture.cc: In constructor ‘GstQSGTexture::GstQSGTexture()’:
/usr/include/glib-2.0/glib/gatomic.h:117:19: error: argument 2 of ‘__atomic_load’ must not be a pointer to a ‘volatile’ type
  117 |     __atomic_load (gapg_temp_atomic, &gapg_temp_newval, __ATOMIC_SEQ_CST); \
      |     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/glib-2.0/glib/gthread.h:260:7: note: in expansion of macro ‘g_atomic_pointer_get’
  260 |     (!g_atomic_pointer_get (location) &&                             \
      |       ^~~~~~~~~~~~~~~~~~~~
/home/acxz/vcs/git/github/acxz/pkgbuilds/qgroundcontrol/src/qgroundcontrol-4.1.3/libs/qmlglsink/gst-plugins-good/ext/qt/gstqsgtexture.cc:42:7: note: in expansion of macro ‘g_once_init_enter’
   42 |   if (g_once_init_enter (&_debug)) {
[... 47 lines cut ...]
/home/acxz/vcs/git/github/acxz/pkgbuilds/qgroundcontrol/src/qgroundcontrol-4.1.3/src/AutoPilotPlugins/PX4/PX4RadioComponent.cc:46:28: error: loop variable ‘mapParam’ of type ‘const QString&’ binds to a temporary constructed from type ‘const char* const’ [-Werror=range-loop-construct]
   46 |         for(const QString &mapParam : {"RC_MAP_ROLL", "RC_MAP_PITCH", "RC_MAP_YAW", "RC_MAP_THROTTLE"}) {
      |                            ^~~~~~~~
/home/acxz/vcs/git/github/acxz/pkgbuilds/qgroundcontrol/src/qgroundcontrol-4.1.3/src/AutoPilotPlugins/PX4/PX4RadioComponent.cc:46:28: note: use non-reference type ‘const QString’ to make the copy explicit or ‘const char* const&’ to prevent copying
cc1plus: all warnings being treated as errors
make: *** [Makefile:126584: PX4RadioComponent.o] Error 1
/home/acxz/vcs/git/github/acxz/pkgbuilds/qgroundcontrol/src/qgroundcontrol-4.1.3/src/AutoPilotPlugins/PX4/PX4AdvancedFlightModesController.cc: In member function ‘void PX4AdvancedFlightModesController::_init()’:
/home/acxz/vcs/git/github/acxz/pkgbuilds/qgroundcontrol/src/qgroundcontrol-4.1.3/src/AutoPilotPlugins/PX4/PX4AdvancedFlightModesController.cc:104:25: error: loop variable ‘attitudeParam’ of type ‘const QString&’ binds to a temporary constructed from type ‘const char* const’ [-Werror=range-loop-construct]
  104 |     for (const QString &attitudeParam : {"RC_MAP_THROTTLE", "RC_MAP_YAW", "RC_MAP_PITCH", "RC_MAP_ROLL", "RC_MAP_FLAPS", "RC_MAP_AUX1", "RC_MAP_AUX2"}) {
      |                         ^~~~~~~~~~~~~
/home/acxz/vcs/git/github/acxz/pkgbuilds/qgroundcontrol/src/qgroundcontrol-4.1.3/src/AutoPilotPlugins/PX4/PX4AdvancedFlightModesController.cc:104:25: note: use non-reference type ‘const QString’ to make the copy explicit or ‘const char* const&’ to prevent copying
/home/acxz/vcs/git/github/acxz/pkgbuilds/qgroundcontrol/src/qgroundcontrol-4.1.3/src/AutoPilotPlugins/PX4/PX4AdvancedFlightModesController.cc: In member function ‘void PX4AdvancedFlightModesController::_validateConfiguration()’:
/home/acxz/vcs/git/github/acxz/pkgbuilds/qgroundcontrol/src/qgroundcontrol-4.1.3/src/AutoPilotPlugins/PX4/PX4AdvancedFlightModesController.cc:173:24: error: loop variable ‘thresholdParam’ of type ‘const QString&’ binds to a temporary constructed from type ‘const char* const’ [-Werror=range-loop-construct]
  173 |     for(const QString &thresholdParam : {"RC_ASSIST_TH", "RC_AUTO_TH", "RC_ACRO_TH", "RC_POSCTL_TH", "RC_LOITER_TH", "RC_RETURN_TH", "RC_OFFB_TH"}) {
      |                        ^~~~~~~~~~~~~~
/home/acxz/vcs/git/github/acxz/pkgbuilds/qgroundcontrol/src/qgroundcontrol-4.1.3/src/AutoPilotPlugins/PX4/PX4AdvancedFlightModesController.cc:173:24: note: use non-reference type ‘const QString’ to make the copy explicit or ‘const char* const&’ to prevent copying
cc1plus: all warnings being treated as errors
  ```
</details>

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.
N/A

acxz, 2021-05-22:
It seems the other error we see in the log (`error: argument 2 of ‘__atomic_load’ must not be a pointer to a ‘volatile’ type`) is solved by the following commit in gst-plugins-good https://gitlab.freedesktop.org/ystreet/gst-plugins-good/-/commit/612102fdbc3f813bf9d3406165692b5f742e51a6

However this commit is not in our fork of gst-plugins-good: https://github.com/mavlink/gst-plugins-good

AndKe, 2022-01-16:
Another one:
```
moc /home/andre/prog/qgroundcontrol/src/MissionManager/CameraSection.h
In file included from /usr/include/glib-2.0/glib/gthread.h:32,
                 from /usr/include/glib-2.0/glib/gasyncqueue.h:32,
                 from /usr/include/glib-2.0/glib.h:32,
                 from /usr/include/gstreamer-1.0/gst/gst.h:27,
                 from /home/andre/prog/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/gstqtglutility.h:24,
                 from /home/andre/prog/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/gstqtglutility.cc:25:
/home/andre/prog/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/gstqtglutility.cc: In function ‘GstGLDisplay* gst_qt_get_gl_display()’:
/usr/include/glib-2.0/glib/gatomic.h:117:19: error: argument 2 of ‘__atomic_load’ must not be a pointer to a ‘volatile’ type
  117 |     __atomic_load (gapg_temp_atomic, &gapg_temp_newval, __ATOMIC_SEQ_CST); \
      |     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/glib-2.0/glib/gthread.h:260:7: note: in expansion of macro ‘g_atomic_pointer_get’
  260 |     (!g_atomic_pointer_get (location) &&                             \
      |       ^~~~~~~~~~~~~~~~~~~~
/home/andre/prog/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/gstqtglutility.cc:65:7: note: in expansion of macro ‘g_once_init_enter’
   65 |   if (g_once_init_enter (&_debug)) {
      |       ^~~~~~~~~~~~~~~~~
make: *** [Makefile:322117: gstqtglutility.o] Error 1
make: *** Waiting for unfinished jobs....
In file included from /usr/include/glib-2.0/glib/gthread.h:32,
                 from /usr/include/glib-2.0/glib/gasyncqueue.h:32,
                 from /usr/include/glib-2.0/glib.h:32,
                 from /usr/include/gstreamer-1.0/gst/gst.h:27,
                 from /usr/include/gstreamer-1.0/gst/video/video.h:23,
                 from /home/andre/prog/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/gstqsgtexture.cc:28:
/home/andre/prog/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/gstqsgtexture.cc: In constructor ‘GstQSGTexture::GstQSGTexture()’:
/usr/include/glib-2.0/glib/gatomic.h:117:19: error: argument 2 of ‘__atomic_load’ must not be a pointer to a ‘volatile’ type
  117 |     __atomic_load (gapg_temp_atomic, &gapg_temp_newval, __ATOMIC_SEQ_CST); \
      |     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/glib-2.0/glib/gthread.h:260:7: note: in expansion of macro ‘g_atomic_pointer_get’
  260 |     (!g_atomic_pointer_get (location) &&                             \
      |       ^~~~~~~~~~~~~~~~~~~~
/home/andre/prog/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/gstqsgtexture.cc:42:7: note: in expansion of macro ‘g_once_init_enter’
   42 |   if (g_once_init_enter (&_debug)) {
      |       ^~~~~~~~~~~~~~~~~
make: *** [Makefile:322227: gstqsgtexture.o] Error 1
In file included from /usr/include/glib-2.0/glib/gthread.h:32,
                 from /usr/include/glib-2.0/glib/gasyncqueue.h:32,
                 from /usr/include/glib-2.0/glib.h:32,
                 from /usr/include/gstreamer-1.0/gst/gst.h:27,
                 from /usr/include/gstreamer-1.0/gst/video/video.h:23,
                 from /home/andre/prog/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/qtwindow.cc:27:
/home/andre/prog/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/qtwindow.cc: In constructor ‘QtGLWindow::QtGLWindow(QWindow*, QQuickWindow*)’:
/usr/include/glib-2.0/glib/gatomic.h:117:19: error: argument 2 of ‘__atomic_load’ must not be a pointer to a ‘volatile’ type
  117 |     __atomic_load (gapg_temp_atomic, &gapg_temp_newval, __ATOMIC_SEQ_CST); \
      |     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/glib-2.0/glib/gthread.h:260:7: note: in expansion of macro ‘g_atomic_pointer_get’
  260 |     (!g_atomic_pointer_get (location) &&                             \
      |       ^~~~~~~~~~~~~~~~~~~~
/home/andre/prog/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/qtwindow.cc:114:7: note: in expansion of macro ‘g_once_init_enter’
  114 |   if (g_once_init_enter (&_debug)) {
      |       ^~~~~~~~~~~~~~~~~
/home/andre/prog/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/qtwindow.cc: In member function ‘void QtGLWindow::beforeRendering()’:
/usr/include/glib-2.0/glib/gatomic.h:117:19: error: argument 2 of ‘__atomic_load’ must not be a pointer to a ‘volatile’ type
  117 |     __atomic_load (gapg_temp_atomic, &gapg_temp_newval, __ATOMIC_SEQ_CST); \
      |     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/glib-2.0/glib/gthread.h:260:7: note: in expansion of macro ‘g_atomic_pointer_get’
  260 |     (!g_atomic_pointer_get (location) &&                             \
      |       ^~~~~~~~~~~~~~~~~~~~
/home/andre/prog/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/qtwindow.cc:160:7: note: in expansion of macro ‘g_once_init_enter’
  160 |   if (g_once_init_enter(&once)) {
      |       ^~~~~~~~~~~~~~~~~
make: *** [Makefile:322667: qtwindow.o] Error 1
In file included from /usr/include/glib-2.0/glib/gthread.h:32,
                 from /usr/include/glib-2.0/glib/gasyncqueue.h:32,
                 from /usr/include/glib-2.0/glib.h:32,
                 from /usr/include/gstreamer-1.0/gst/gst.h:27,
                 from /usr/include/gstreamer-1.0/gst/video/video.h:23,
                 from /home/andre/prog/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/qtitem.cc:27:
[... cut ...]
      |     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/glib-2.0/glib/gthread.h:260:7: note: in expansion of macro ‘g_atomic_pointer_get’
  260 |     (!g_atomic_pointer_get (location) &&                             \
      |       ^~~~~~~~~~~~~~~~~~~~
/home/andre/prog/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/qtitem.cc:111:7: note: in expansion of macro ‘g_once_init_enter’
  111 |   if (g_once_init_enter (&_debug)) {
      |       ^~~~~~~~~~~~~~~~~
make: *** [Makefile:322843: qtitem.o] Error 1

```

AndKe, 2022-01-16:
@acxz is there a workaround for this? - can I specify an older gcc?

AndKe, 2022-01-16:
@DonLakeFlyer I am forced to ping you too on this, please help make it possible to build the current master.

bobosito000, 2022-01-19:
I made this pkgbuild to build on my system. It applies the patch and compiles fine on my computer. Not sure if this is the type of workaround you are looking for, but here's the repo [https://github.com/bobosito000/arch-qgroundcontrol/blob/main/PKGBUILD](https://github.com/bobosito000/arch-qgroundcontrol/blob/main/PKGBUILD).

AndKe, 2022-01-19:
This is strange, but on two Ubuntu systems, (same version) , it seems that only one fails with the kind of errors, the other one compiles it just fine, and the result runs just fine on both.

bobosito000, 2022-01-19:
That's strange. I'm not sure how you are running the pkgbuild on ubuntu. I've never done that before. Are you manually running the shell commands in the pkgbuild functions?

AndKe, 2022-01-19:
@bobosito000  no, I did not even try the packagebuild, I refer to the normal qmake / make build procedure.
The only thing I absolutely fail to build is when I add libgstreamer1.0-dev  to enable video support. (the perquisites are not very well described on this topic, but adding that library tries to build with video.)
I believe this is a known issue?

bobosito000, 2022-01-19:
Yeah. If I were you, I would try downloading the patch file from my repo. Put the patch file into your qgroundcontrol repository folder and run the following:
```
patch -p1 -i patch
cmake . -B ./build
cd build
make -j3
```
I'm pretty sure it would work on your system.

bobosito000, 2022-01-19:
Errors without the patch:
```
[  1%] Building CXX object libs/qmlglsink/CMakeFiles/qmlglsink.dir/gst-plugins-good/ext/qt/gstqtglutility.cc.o
In file included from /usr/include/glib-2.0/glib/gthread.h:32,
                 from /usr/include/glib-2.0/glib/gasyncqueue.h:32,
                 from /usr/include/glib-2.0/glib.h:32,
                 from /usr/include/gstreamer-1.0/gst/gst.h:27,
                 from /home/bobosito/tmp/qgroundcontrol/src/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/gstqtglutility.h:24,
                 from /home/bobosito/tmp/qgroundcontrol/src/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/gstqtglutility.cc:25:
/home/bobosito/tmp/qgroundcontrol/src/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/gstqtglutility.cc: In function ‘GstGLDisplay* gst_qt_get_gl_display()’:
/usr/include/glib-2.0/glib/gatomic.h:113:19: error: argument 2 of ‘__atomic_load’ must not be a pointer to a ‘volatile’ type
  113 |     __atomic_load (gapg_temp_atomic, &gapg_temp_newval, __ATOMIC_SEQ_CST); \
      |     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/glib-2.0/glib/gthread.h:260:7: note: in expansion of macro ‘g_atomic_pointer_get’
  260 |     (!g_atomic_pointer_get (location) &&                             \
      |       ^~~~~~~~~~~~~~~~~~~~
/home/bobosito/tmp/qgroundcontrol/src/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/gstqtglutility.cc:65:7: note: in expansion of macro ‘g_once_init_enter’
   65 |   if (g_once_init_enter (&_debug)) {
      |       ^~~~~~~~~~~~~~~~~
make[2]: *** [libs/qmlglsink/CMakeFiles/qmlglsink.dir/build.make:104: libs/qmlglsink/CMakeFiles/qmlglsink.dir/gst-plugins-good/ext/qt/gstqtglutility.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1632: libs/qmlglsink/CMakeFiles/qmlglsink.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
```

bobosito000, 2022-01-19:
after applying patch:
```
[bobosito@oryx qgroundcontrol]$ patch -p1 -i patch
patching file libs/qmlglsink/gst-plugins-good/ext/qt/gstqsgtexture.cc
patching file libs/qmlglsink/gst-plugins-good/ext/qt/gstqtglutility.cc
patching file libs/qmlglsink/gst-plugins-good/ext/qt/qtitem.cc
patching file libs/qmlglsink/gst-plugins-good/ext/qt/qtwindow.cc
[bobosito@oryx qgroundcontrol]$ cd build/
[bobosito@oryx build]$ make
[  0%] Automatic MOC and UIC for target qmlglsink
[  0%] Built target qmlglsink_autogen
Consolidate compiler generated dependencies of target qmlglsink
[  1%] Building CXX object libs/qmlglsink/CMakeFiles/qmlglsink.dir/gst-plugins-good/ext/qt/gstqtglutility.cc.o
[  1%] Building CXX object libs/qmlglsink/CMakeFiles/qmlglsink.dir/gst-plugins-good/ext/qt/gstqsgtexture.cc.o
[  1%] Building CXX object libs/qmlglsink/CMakeFiles/qmlglsink.dir/gst-plugins-good/ext/qt/gstqtsrc.cc.o
[  1%] Building CXX object libs/qmlglsink/CMakeFiles/qmlglsink.dir/gst-plugins-good/ext/qt/qtwindow.cc.o
^[[115;10u[  2%] Building CXX object libs/qmlglsink/CMakeFiles/qmlglsink.dir/gst-plugins-good/ext/qt/qtitem.cc.o
[  2%] Linking CXX static library libqmlglsink.a
[  2%] Built target qmlglsink
[  2%] Automatic MOC and UIC for target VideoReceiver
[  2%] Built target VideoReceiver_autogen
[  3%] Automatic MOC and UIC for target libevents_generated
[  3%] Built target libevents_generated_autogen
```

AndKe, 2022-01-19:
@bobosito000  Thank you.
I will need to test this at home too..
I build master, after installing libgstreamer1.0-dev  - qmake confirms that video is enabled.
then the build just builds w/o errors.
but I see no video options, the topmost droplist box have no more items.
[image]

This was to be my "before patch" test, I expected the build to fail.

AndKe, 2022-01-20:
@bobosito000 I tried to compile at home, and (unlike the other computers) ran into the errors.
Applied your patch - and it works fine.
Also, the source droplist is properly populated when building on this computer,  (but I believe this has something to do with incomplete perquisites on the other computers  - as mentioned in https://github.com/mavlink/qgroundcontrol/issues/10118)

I hope you created a PR for that patch? - it's clearly great.

DonLakeFlyer, 2022-01-20:
> I am forced to ping you too on this, please help make it possible to build the current master.

The video stuff is mostly a mystery to me. Someone needs to take over ownership of it and make it work for real. At this point I'm no longer in charge of merging in pulls.

AndKe, 2022-01-20:
@acxz you had some suggestions here https://github.com/mavlink/gst-plugins-good/pull/1#issuecomment-928415712
please look into it.

AndKe, 2022-01-20:
@DonLakeFlyer - in any case, patching current master / accepting that PR, can't have any adverse effects.

smagellan, 2022-01-21:
@DonLakeFlyer, https://github.com/mavlink/gst-plugins-good/pull/1#issuecomment-1003546627.
You can try vanilla version as temporary workaround:
```bash
#! /bin/bash

set -euxo pipefail;
VANILLA_GST_PLUGINS_DIFF="$PWD/qgroundcontrol_vanilla_gst_plugins_good_1_18.diff";
git clone --recursive -j8 --branch "Stable_V4.2" https://github.com/mavlink/qgroundcontrol;
cd qgroundcontrol;
git fetch --all --tags -f;
git submodule set-url libs/qmlglsink/gst-plugins-good https://github.com/GStreamer/gst-plugins-good.git;
git submodule set-branch --branch 1.18 libs/qmlglsink/gst-plugins-good;
git submodule sync
```
qgroundcontrol_vanilla_gst_plugins_good_1_18.diff adds 2 more gst-specific files (were added to upstream) into libs/qmlglsink/CMakeLists.txt and qmlglsink.pri

smagellan, 2022-01-21:
Did you build it within source directory? Looks like it resets to initial gst if yes. 
P.S. I updated bash script

bobosito000, 2022-01-21:
Thanks for making me aware about the git submodules being automatically updated. Tried your script out and make -j3 in your qgroundcontrol-build folder.  Still gives me the volatile errors.

<details>
  <summary>make errors:</summary>
  <p>

```
compiling /home/bobosito/lib/tools/qgroundcontrol/src/VideoManager/VideoManager.cc
compiling /home/bobosito/lib/tools/qgroundcontrol/src/VideoReceiver/gstqgcvideosinkbin.c
/home/bobosito/lib/tools/qgroundcontrol/src/VideoReceiver/gstqgcvideosinkbin.c: In function ‘_vsb_get_type’:
/home/bobosito/lib/tools/qgroundcontrol/src/VideoReceiver/gstqgcvideosinkbin.c:314:13: warning: cast between incompatible function types from ‘void (*)(GstQgcVideoSinkBinClass *)’ {aka ‘void (*)(struct _GstQgcVideoSinkBinClass *)’} to ‘void (*)(void *, void *)’ [-Wcast-function-type]
  314 |             (GClassInitFunc)_vsb_class_init,
      |             ^
/home/bobosito/lib/tools/qgroundcontrol/src/VideoReceiver/gstqgcvideosinkbin.c:319:13: warning: cast between incompatible function types from ‘void (*)(GstQgcVideoSinkBin *)’ {aka ‘void (*)(struct _GstQgcVideoSinkBin *)’} to ‘void (*)(GTypeInstance *, void *)’ {aka ‘void (*)(struct _GTypeInstance *, void *)’} [-Wcast-function-type]
  319 |             (GInstanceInitFunc)_vsb_init,
      |             ^
compiling /home/bobosito/lib/tools/qgroundcontrol/src/VideoReceiver/gstqgc.c
compiling /home/bobosito/lib/tools/qgroundcontrol/src/VideoReceiver/GStreamer.cc
compiling /home/bobosito/lib/tools/qgroundcontrol/src/VideoReceiver/GstVideoReceiver.cc
compiling /home/bobosito/lib/tools/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/gstplugin.cc
compiling /home/bobosito/lib/tools/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/gstqtglutility.cc
compiling /home/bobosito/lib/tools/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/gstqsgtexture.cc
In file included from /usr/include/glib-2.0/glib/gthread.h:32,
                 from /usr/include/glib-2.0/glib/gasyncqueue.h:32,
                 from /usr/include/glib-2.0/glib.h:32,
                 from /usr/include/gstreamer-1.0/gst/gst.h:27,
                 from /home/bobosito/lib/tools/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/gstqtglutility.h:24,
                 from /home/bobosito/lib/tools/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/gstqtglutility.cc:25:
/home/bobosito/lib/tools/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/gstqtglutility.cc: In function ‘GstGLDisplay* gst_qt_get_gl_display()’:
/usr/include/glib-2.0/glib/gatomic.h:113:19: error: argument 2 of ‘__atomic_load’ must not be a pointer to a ‘volatile’ type
  113 |     __atomic_load (gapg_temp_atomic, &gapg_temp_newval, __ATOMIC_SEQ_CST); \
      |     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/glib-2.0/glib/gthread.h:260:7: note: in expansion of macro ‘g_atomic_pointer_get’
  260 |     (!g_atomic_pointer_get (location) &&                             \
      |       ^~~~~~~~~~~~~~~~~~~~
/home/bobosito/lib/tools/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/gstqtglutility.cc:65:7: note: in expansion of macro ‘g_once_init_enter’
   65 |   if (g_once_init_enter (&_debug)) {
      |       ^~~~~~~~~~~~~~~~~
make: *** [Makefile:160753: gstqtglutility.o] Error 1
make: *** Waiting for unfinished jobs....
In file included from /usr/include/glib-2.0/glib/gthread.h:32,
                 from /usr/include/glib-2.0/glib/gasyncqueue.h:32,
                 from /usr/include/glib-2.0/glib.h:32,
                 from /usr/include/gstreamer-1.0/gst/gst.h:27,
                 from /usr/include/gstreamer-1.0/gst/video/video.h:23,
                 from /home/bobosito/lib/tools/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/gstqsgtexture.cc:28:
/home/bobosito/lib/tools/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/gstqsgtexture.cc: In constructor ‘GstQSGTexture::GstQSGTexture()’:
/usr/include/glib-2.0/glib/gatomic.h:113:19: error: argument 2 of ‘__atomic_load’ must not be a pointer to a ‘volatile’ type
  113 |     __atomic_load (gapg_temp_atomic, &gapg_temp_newval, __ATOMIC_SEQ_CST); \
      |     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/glib-2.0/glib/gthread.h:260:7: note: in expansion of macro ‘g_atomic_pointer_get’
  260 |     (!g_atomic_pointer_get (location) &&                             \
      |       ^~~~~~~~~~~~~~~~~~~~
/home/bobosito/lib/tools/qgroundcontrol/libs/qmlglsink/gst-plugins-good/ext/qt/gstqsgtexture.cc:42:7: note: in expansion of macro ‘g_once_init_enter’
   42 |   if (g_once_init_enter (&_debug)) {
      |       ^~~~~~~~~~~~~~~~~
make: *** [Makefile:160758: gstqsgtexture.o] Error 1
[bobosito@oryx qgroundcontrol-build]$
```
</p>
</details>

Regardless though, qgroundcontrol's master branch builds fine now. The latest mavlink gst plugins merge fixes the issue.

smagellan, 2022-01-22:
@bobosito000 sorry, I have no other guesses

acxz, 2022-01-25:
Closing this issue as the associated PR (https://github.com/mavlink/qgroundcontrol/pull/10132) has been merged.

closed 2022-01-25 by acxz