mavlink/qgroundcontrol #9678: Compile error: libevents/libs/cpp/parse/parser.h: No such file or directory

heeplr, 2021-05-25:
## Expected Behavior
Successful build.

## Current Behavior
```
In file included from /var/tmp/portage/app-misc/qgroundcontrol-9999/work/qgroundcontrol-9999/src/Vehicle/EventHandler.h:18,
                 from /var/tmp/portage/app-misc/qgroundcontrol-9999/work/qgroundcontrol-9999/src/Vehicle/Vehicle.cc:52:
/var/tmp/portage/app-misc/qgroundcontrol-9999/work/qgroundcontrol-9999/src/Vehicle/HealthAndArmingChecks.h:16:10: fatal error: libevents/libs/cpp/parse/parser.h: No such file or directory
```

## System Information
- Operating System: Gentoo Linux amd64
- QGC Version: current master
- QGC build: -

## Log Files and Screenshots
s. attached [build.log](https://github.com/mavlink/qgroundcontrol/files/6539405/build.log)

WTPENGUIN, 2021-05-25:
Try this command in the source code folder.

git submodule update --init --recursive

heeplr, 2021-05-26:
I double checked the source tree and the file is there. It's just not found.

The compile command doesn't seem to contain the needed include path. The only include paths inside "libs" are:
* /libs/eigen 
* /libs/mavlink/include/mavlink/v2.0
* /libs/mavlink/include/mavlink/v2.0/ardupilotmega
* /libs/mavlink/include/mavlink/v2.0/common
* /libs/shapelib

The problem is probably caused by missing cmake integration (?) and triggered when building outside of the sourcetree. It should be reproducable with:
```
$ git clone --recursive -j8 https://github.com/mavlink/qgroundcontrol.git
$ cd qggroundcontrol
$ git submodule update --recursive
$ mkdir build
$ cd build
$ cmake ..
$ make
```

DonLakeFlyer, 2021-05-26:
Change the Events submodule INCLUDEPATH in QGCExternalLibs.pri to this:
```
INCLUDEPATH += \
        libs/libevents \
        libs/libevents/libs/cpp/parse
```

We're still not exactly sure why that is needed since it is a relative include. But it fixes it. I'll make a pull for this today.

heeplr, 2021-05-30:
Does that fix it for you? Neither this nor the actual path ("libs/libevents/libevents/libs/cpp/parse") does for me.

When compiling with ```VERBOSE=1 make```, I don't even see the "-I libs/libevents" added to the compile command (which should enable the compiler to resolve the "#include <libevents/libs/cpp/parse/parser.h>" statement in the first place.)
So my changes aren't added aswell.

I can't find a reason why that isn't included but the following "libs/shapelib" is. I doubt it but maybe it's a cmake bug?
(cmake version 3.18.5 and g++-11.1.0 p1 here btw)

heeplr, 2021-05-30:
I found the reason why shapelib include paths are added but libevents' not. I'll create a PR.

DonLakeFlyer, 2021-05-30:
qmake is the officially supported version for building. cmake has partial support and breaks from time to time.

fpemud, 2022-07-01:
hi, @heeplr 
It's great to see that you are building qgroundcontrol on Gentoo Linux. 
Could you share your ebuild file (app-misc/qgroundcontrol-9999)?

heeplr, 2022-07-01:
Hi @fpemud 

Here's my ebuild, but it's hardly fit to release as it's broken atm. QGroundcontrol compiles successfully but when launched, can't find libshp.so.1 and some other libraries/gstreamer modules that are needed. I didn't find the time to address it, yet :-(

```
# Copyright 1999-2018 Gentoo Foundation
# Distributed under the terms of the GNU General Public License v2

EAPI=7

inherit qmake-utils desktop flag-o-matic git-r3

DESCRIPTION="ground control station for drones"
HOMEPAGE="http://qgroundcontrol.io"
EGIT_REPO_URI="https://github.com/mavlink/${PN}.git"

LICENSE="Apache-2.0 GPL-3"
SLOT="0"
KEYWORDS="~amd64"
IUSE=""

DEPEND="dev-qt/qtbluetooth:5
        dev-qt/qtcharts:5[qml]
        dev-qt/qtconcurrent:5
        dev-qt/qtcore:5
        dev-qt/qtx11extras:5
        dev-qt/qtgraphicaleffects:5
        dev-qt/qtlocation:5
        dev-qt/qtmultimedia:5[qml]
        dev-qt/qtnetwork:5
        dev-qt/qtpositioning:5[qml]
        dev-qt/qtquickcontrols2:5[widgets]
        dev-qt/qtspeech:5
        dev-qt/qtspeech:5
        dev-qt/qtsql:5
        dev-qt/qtsvg:5
        dev-qt/qttest:5
        dev-qt/qtwidgets:5
        dev-qt/qtxml:5
        sci-libs/shapelib"
RDEPEND="${DEPEND}"

pkg_setup() {
        append-cxxflags \
                -Wno-error=int-in-bool-context \
                -Wno-error=ignored-attributes \
                -Wno-error=misleading-indentation \
                -Wno-error=implicit-fallthrough
}

src_prepare() {
        #bundled mavlink, eigen, maybe others
        default
        #sed -e "/^Exec=/s:qgroundcontrol-start.sh:/usr/bin/${PN}:" \
        #       -i deploy/${PN}.desktop
}

src_install() {
        newbin "${WORKDIR}/${P}/build/gcc_64/QGroundControl" ${PN}
        doicon resources/icons/${PN}.png
        domenu deploy/${PN}.desktop
}
```

chron0, 2023-07-04:
any update on thise one? I still see this issue when bumping the ebuild to 4.2.6

chron0, 2023-07-04:
hmkay, so it looks like they are built:

```
chronos /var/tmp/portage/app-misc/qgroundcontrol-4.2.6 # find . -name libshp.so.1
./work/qgroundcontrol-4.2.6/build/gcc_64/libs/shapelib/libshp.so.1
chronos /var/tmp/portage/app-misc/qgroundcontrol-4.2.6 # find . -name libqmlglsink.so
./work/qgroundcontrol-4.2.6/build/gcc_64/libs/qmlglsink/libqmlglsink.so
```

@heeplr do you have any idea how we tell doins to copy them over from the build sandbox into the system during install phase? that may be all that is missing here.

chron0, 2023-07-04:
@heeplr wanna try this one?

```
# Copyright 1999-2018 Gentoo Foundation
# Distributed under the terms of the GNU General Public License v2

EAPI=8

inherit qmake-utils desktop flag-o-matic git-r3

DESCRIPTION="ground control station for drones"
HOMEPAGE="http://qgroundcontrol.io"
EGIT_REPO_URI="https://github.com/mavlink/${PN}.git"

LICENSE="Apache-2.0 GPL-3"
SLOT="0"
KEYWORDS="~amd64"
IUSE=""

DEPEND="dev-qt/qtbluetooth:5
        dev-qt/qtcharts:5[qml]
        dev-qt/qtconcurrent:5
        dev-qt/qtcore:5
        dev-qt/qtx11extras:5
        dev-qt/qtgraphicaleffects:5
        dev-qt/qtlocation:5
        dev-qt/qtmultimedia:5[qml]
        dev-qt/qtnetwork:5
        dev-qt/qtpositioning:5[qml]
        dev-qt/qtquickcontrols2:5[widgets]
        dev-qt/qtspeech:5
        dev-qt/qtspeech:5
        dev-qt/qtsql:5
        dev-qt/qtsvg:5
        dev-qt/qttest:5
        dev-qt/qtwidgets:5
        dev-qt/qtxml:5
        sci-libs/shapelib"
RDEPEND="${DEPEND}"

pkg_setup() {
        append-cxxflags \
                -Wno-error=int-in-bool-context \
                -Wno-error=ignored-attributes \
                -Wno-error=misleading-indentation \
                -Wno-error=implicit-fallthrough
}

src_prepare() {
        #bundled mavlink, eigen, maybe others
        default
        #sed -e "/^Exec=/s:qgroundcontrol-start.sh:/usr/bin/${PN}:" \
        #       -i deploy/${PN}.desktop
}

src_install() {
        newbin "${WORKDIR}/${P}/build/gcc_64/QGroundControl" ${PN}
        newlib.so "${WORKDIR}/${P}/build/gcc_64/libs/shapelib/libshp.so.1.5.0" libshp.so.1.5.0 
        newlib.so "${WORKDIR}/${P}/build/gcc_64/libs/qmlglsink/libqmlglsink.so" libqmlglsink.so 
        dosym /usr/lib64/libshp.so.1.5.0 /usr/lib64/libshp.so.1
        doicon resources/icons/${PN}.png
        domenu deploy/${PN}.desktop
}
```

chron0, 2023-07-04:
i am just a little confused why it tags itself as v4.2.0 in the UI when pulling from the repo like that...

chron0, 2023-07-04:
looking good so far:

[image]